sav-oss-eplanner/EasyEPLANner #28: Добавить механизм версий

dimabeel, 2019-12-27:
**Ваш запрос связан с проблемой? Пожалуйста, опишите.**
Нет возможности узнать версию системы.

**Опишите решение, которое Вы хотели бы**
Добавить механизм версий для надстройки, что бы лучше ориентироваться в ней при возникновении каких либо ошибок, или при внесении доработок.

**Опишите альтернативы, которые Вы рассматривали**
Других альтернатив в надстройке нет.

**Дополнительная информация**
Надо проработать систему версий, её нумерацию и порядок.

dimabeel, 2019-12-27:
@idzm, есть предложения по нумерации и порядку этой нумерации? Не учитывая имеющуюся по типу *год.месяц*. Если в месяце несколько версий выпущено, то надо что-то ещё. Например, *год.месяц.день*. А если за 1 день несколько багфиксов, то предусмотреть *год.месяц.день.порядковый_номер_за_день*?

dimabeel, 2019-12-30:
Данное предложение  принято при обсуждении с @idzm.
> год.месяц.день.порядковый_номер_за_день

idzm, 2019-12-30:
Может такой вариант:  `год.месяц.день.глобальный_порядковый_номер_сборки`?

dimabeel, 2019-12-30:
Принят вариант: `год.месяц.сборка.ревизия`. В итоге можно получить, например: _2019.13.7303.15000_.

Поля **Сборка** и **Ревизия** вычисляются системой автоматически.

**Сборка** - день, отсчитываемый с 1 _января_ 2000 года. 
**Ревизия** - время в секундах, отсчитываемое с полуночи текущего дня.

idzm, 2019-12-30:
Согласен, принято.

PR #29 (2020-01-03): Added button about add-in version
Fixes #28.

*Предлагается добавить этот функционал вместе со следующим серьезным функционалом*
closed 2020-01-03 by idzm