sav-oss-eplanner/EasyEPLANner #30: Привязка дискретных устройств к модулям IO-Link

KUZMA-1989, 2019-12-30:
Все дискретные устройства должны иметь возможность привязываться к IO-Link каналам.

dimabeel, 2019-12-30:
@KUZMA-1989 , по принципу 1 устройство - 1 канал (DI или DO) IO-Link?

dimabeel, 2019-12-30:
@KUZMA-1989 , сигналы DI,DO учитывать (которые вне устройств)? Устройства - дискретный выход, вход

KUZMA-1989, 2019-12-31:
Да,любое устройство или дискретный сигнал DI-DO можно подключить к каналу IO-Link

dimabeel, 2019-12-31:
@KUZMA-1989 , предполагается, что устройство имеет только 1 сигнал DI или DO? Или же возможно наличие и других сигналов (AI,AO, а так же смесь аналоговых и дискретных сигналов)

KUZMA-1989, 2019-12-31:
У модуля IO-Link канал настроить можно дополнительно на DO или DI, AI и AO нету

dimabeel, 2019-12-31:
@KUZMA-1989 , я про привязываемые устройства. Привязываемое устройство какой набор каналов может иметь?

KUZMA-1989, 2019-12-31:
[image]

KUZMA-1989, 2019-12-31:
@dimabeel ТЫ про это?

dimabeel, 2019-12-31:
@KUZMA-1989 , нет. Количество каналов (сигналов) на устройство и их типы. Например, двигатель M может иметь канал AO и канал DI. Такие комбинации предполагают же, что DI привяжется на IO-Link, а AO в другое место?

KUZMA-1989, 2019-12-31:
Да

PR #33 (2020-01-13): Added binding DI&DO to IO-Link
Fixes #30.
Fixes #31.

**В работе**

Необходимо переписать алгоритм привязки полностью, а так же изменится концепция работы с модулями ввода-вывода.

Для обозначения, к какому каналу привязывать (DI,DO, AI/AO) будет использоваться дополнительная подпись в функциональном тексте устройств, которые привязываются к IO-Link.
closed 2020-01-13 by skiftol