sav-oss-eplanner/EasyEPLANner #31: Выбор тип канала на модулях IO-Link

KUZMA-1989, 2019-12-30:
Для того, чтобы исключить ошибку привязки, для каждого канала добавляется выбор типа: 
* DI; 
* DO; 
* IO-Link;
* отключен.

dimabeel, 2019-12-30:
@KUZMA-1989 , что должен делать тип "Отключен"? Он чисто графически отображается? Или будет привязано устройство, но отключено?

dimabeel, 2019-12-30:
@KUZMA-1989 , 

> для каждого канала добавляется выбор типа:

Для каждого канала **IO-Link** модуля?

KUZMA-1989, 2019-12-31:
Честно я не знаю как это реализовать, это @idzm захотел

dimabeel, 2019-12-31:
@KUZMA-1989 , Ок. Буду работать

dimabeel, 2020-01-09:
@idzm, реализация типа канала без типа "Отключен" пойдет? Я не вижу никакой смысловой нагрузки, потому что если к каналу ничего не привязано, то в проектах много где пишут "Резерв", что бы использовать канал как резервный в будущем, или ничего не пишут.

А если к каналу что-то привязано, то оно в работе.

@KUZMA-1989 , я все верно описал?

idzm, 2020-01-09:
Ок, может быть и так.

PR #33 (2020-01-13): Added binding DI&DO to IO-Link
Fixes #30.
Fixes #31.

**В работе**

Необходимо переписать алгоритм привязки полностью, а так же изменится концепция работы с модулями ввода-вывода.

Для обозначения, к какому каналу привязывать (DI,DO, AI/AO) будет использоваться дополнительная подпись в функциональном тексте устройств, которые привязываются к IO-Link.
closed 2020-01-13 by skiftol