sav-oss-eplanner/EasyEPLANner #59: Добавить датчик давления FESTO SPAE с IO-Link

KUZMA-1989, 2020-01-16:
Добавить датчик давления FESTO SPAE с IO-Link  в eplan. Основное обозначение в eplan будет PT, доп поле 2: DEV_SPAE
[100017.pdf](https://github.com/savushkin-r-d/EasyEPLANner/files/4070869/100017.pdf)
DEV_SPAE

idzm, 2021-09-20:
Дополнительное поле оставляем как у обычного IO-Link датчика - `DST_PT_IOLINK`. Для отличия от других, в графе артикула пишем `FESTO.SPAE-P10R`.

dimabeel, 2021-09-22:
> Дополнительное поле оставляем как у обычного **IO-Link** датчика - `DST_PT_IOLINK`. Для отличия от других в графе артикула пишем `FESTO.SPAE-P10R`.

Не `DST_PT_IOLINK`, а `PT_IOLINK`. Изделие остается прежним т.к в таком виде оно было взято с Data Portal EPLAN. Дополнительно надо сдвинуть подтип `PT_VIRT` с перечислении с **4** позиции на **3** (_нигде не использовалось, допустимо_).

PR #60 (2020-01-23): Added PT DEV_SPAE device subtype
Fixes #59

**Решение**
Добавлен датчик давления воздуха PT с подтипом DEV_SPAE, обновлена документация.
PR #913 (2021-09-22): Move DEV_SPAE to PT_IOLINK
Fixes #59
closed 2021-09-22 by idzm