sav-oss-eplanner/EasyEPLANner #73: Дополнительные параметры для размеров танка для датчиков LT_IOLINK

KUZMA-1989, 2020-01-25:
Добавить варианты датчиков LT с IO-Link интерфейсом LT_IOLINK_CYL и параметры (P_MAX_P, P_R), LT_IOLINK_CONE параметры (P_MAX_P, P_R, P_H_CONE), LT_IOLINK_TRUNC параметры (P_MAX_P, P_R, P_H_TRUNC), чтоб велся автоматический пересчет в программе данных в литры продукта, реализация аналогична старым датчикам без IO-Link.

idzm, 2020-01-25:
@KUZMA-1989 - предлагаю все эти параметры добавить в текущее устройство LT_IOLINK. Это будет проще и удобнее в использовании.

dimabeel, 2020-01-25:
@idzm, а как тогда сигнализировать об ошибке? Если, например задано 2 параметра - это LT_IOLINK_CYL. Окей, это можно проверить. А вот если должно быть задано 3 параметра? Система не посчитает ошибку, потому что два из трех параметров будут присутствовать, и подтип будет соответствовать LT_IOLINK_CYL и инженер по АСУТП не заметит ошибки. Причем отловить это будет не так уж и легко.

idzm, 2020-01-25:
Надо, чтобы были заданы все параметры. Те, которые не нужны для расчёта, будут иметь значение ноль - 0.

dimabeel, 2020-01-25:
То есть, строка параметров для:
1. _LT_IOLINK_CYL:_ "P_MAX_P={**value**}, P_R={**value**}, P_H_CONE=**0**, P_H_TRUNC=**0**";
2. _LT_IOLINK_CONE_: "P_MAX_P={**value**}, P_R={**value**}, P_H_CONE={**value**}, P_H_TRUNC=**0**";
3. _LT_IOLINK_CYL_: "P_MAX_P={**value**}, P_R={**value**}, P_H_CONE=**0**, P_H_TRUNC={**value**}";

Если так, то можно. Вопрос только, понравится этот вариант @KUZMA-1989 и остальным.

KUZMA-1989, 2020-01-25:
Так мы же сами в доп поле 3 указываем эти параметры, и в зависимости какой тип датчика указывали, те параметры и прописывали, а если какой-то отсутствовал, то появлялась ошибка. А сейчас нужно будет прописать все подряд параметры и значения указать нужные, а те, которые не нужны поставить 0 ?

idzm, 2020-01-25:
Тип датчика также будет параметром.

dimabeel, 2020-01-25:
Какой-то разрыв шаблона. Все датчики указывались, как и указываются, а вот для LT_IOLINK надо будет выделываться с параметрами. Если я правильно понял, то появится параметр какой-нибудь типа "TYPE=" который сможет принимать значения "CYL, CONE, TRUNC" конкретно для LT_IOLINK.

Если @idzm предлагает нечто похожее, тогда надо для всех IO-Link датчиков использовать такой вариант (если например PT_IOLINK будет иметь какие-то особенные типы, как с DEV_SPAE сейчас).

KUZMA-1989, 2020-01-25:
У нас уже было такие устройства только без IOLINK, я хочу чтоб этот функционал работал

KUZMA-1989, 2020-01-25:
@dimabeel Посмотри как раньше работала логика для датчиков LT

dimabeel, 2020-01-25:
В епланере логики для датчиков нет, вся логика в управляющей программе.

KUZMA-1989, 2020-01-25:
Тогда разбирайтесь между собой как вернуть старую логику в LT

dimabeel, 2020-01-25:
Хорошо, в понедельник решим

dimabeel, 2020-03-05:
@idzm , нужно решить этот вопрос как можно скорее

idzm, 2020-03-09:
В целях упрощения предлагается такой общий тип: `LT_IOLINK_TANK`.
Cтрока параметров будет такая:
```lua
"P_MAX_P={value}, P_R={value}, P_H_CONE={value or 0}, P_H_TRUNC={value or 0}"
```

dimabeel, 2020-03-10:
> В целях упрощения предлагается такой общий тип: `LT_IOLINK_TANK`.
> Cтрока параметров будет такая:
> 
> ```lua
> "P_MAX_P={value}, P_R={value}, P_H_CONE={value or 0}, P_H_TRUNC={value or 0}"
> ```

Лучше использовать `LT_IOLINK`, но в параметрах задавать:

`"P_MAX_P={value}, P_R={value}, P_H_CYL ={value or 0} P_H_CONE={value or 0}, P_H_TRUNC={value or 0}"`

Или же попробовать вариант без задания P_H_CYL, P_H_CONE, P_H_TRUNC, а задавать их только в случае, если это необходимо. То есть, один из этих параметров появится только если мы будем указывать какой-то определенный тип LT_IOLINK. Если задано несколько из этих параметров, то выдавать ошибку и всё

idzm, 2020-03-10:
Так и решаем.

idzm, 2020-03-10:
Упрощаем строку параметров:

"P_MAX_P={value}, P_R={value}, P_H_CONE={value or 0}"

PR #178 (2020-03-10): Added LT_IOLINK parameters for different tanks types
Fixes #73.

**Решение**
Добавлены параметры для LT_IOLINK.
Теперь для разных типов танка мы попросту задаем параметры в LT_IOLINK.
https://github.com/savushkin-r-d/EasyEPLANner/tree/master/docs/user_manual#532-Назначение-дополнительных-полей - здесь документация.
closed 2020-03-10 by idzm