sav-oss-eplanner/EasyEPLANner #93: Заменить текущую библиотеку для работы с Excel на EPPlus

dimabeel, 2020-01-29:
Необходимо заменить текущую библиотеку для работы с Excel (генерация документа из ExcelReporter) на OSP библиотеку EPPlus. В проект должен быть добавлен subModule в виде этой библиотеки. А подключаться библиотека должна из папки libs (просто скопировать туда файлик dll библиотеки).  Также надо учесть, что библиотека dll должна иметь строгое имя (быть подписана). Можно собственным ключом, или ключом от разработчика (может быть подписано разработчиком сразу).

AS005016, 2020-01-29:
Попробую разбить на подзадачи  и делать последовательно.

dimabeel, 2020-01-30:
Начать можно с класса ExportTechDevsToExcel в main. Это запуск экспорта. Таким образом можно разобраться как он работает. Для начала надо разобраться с текущим алгоритмом работы, а после - начинать работу по переделыванию

dimabeel, 2020-02-10:
В данный момент модифицируется экспорт в Excel (#121). Принимайте эти изменения к сведению, что бы функционал не отличался.

AS005016, 2020-02-13:
Изучаю текущую библиотеку работы с Excel. Части кода в которых осуществляется работа с Excel (классы и их функциональность).
[report.txt](https://github.com/savushkin-r-d/EasyEPLANner/files/4198214/report.txt)

dimabeel, 2020-02-13:
> Изучаю текущую библиотеку работы с Excel. Части кода в которых осуществляется работа с Excel (классы и их функциональность).
> [report.txt](https://github.com/savushkin-r-d/EasyEPLANner/files/4198214/report.txt)

То, что написано в report.txt можно было скопировать сюда. Лучше, что бы было так. Не надо тогда ничего качать и смотреть

AS005016, 2020-02-13:
**Выполнение задания начала с изучения текущего алгоритма взаимодействия с Excel.** 

1. ExportTechDevsToExcel- экспорт технических разработок в Excel
ActionCallingContext- контекст вызванного действия.
SaveAsExcelDescription - сохранение Excel описания.
В ExcelRepoter начинается основная работа с Excel: 
- Создание 
- Сохранение Excel файла 
- Заполнение ячеек 
- Выравнивание
- Создание страницы с модулями IO
- Форматирование страницы (Форматирование по ширине содержимого)
- Выделение отдельным блоком модулей AS-interface
- Создание сводной таблицы используемых модулей на основании словаря
- Заполнение таблицы
- Окрас ячеек
- Создание страницы с устройствами для операций и шагов технических объектов
- Заполнение страницы данными
- Установка переноса текста в ячейке
- Создание страницы с параметрами технических объектов проекта
- Создание страницы с описанием устройств
- Настройка имен столбцов и так далее.
2. Изучение библиотеки EPPluse.
[https://riptutorial.com/ru/epplus](url)

AS005016, 2020-02-25:
**3. Ознакомилась с библиотекой EPPlus:**
   Изучила:
- Лицензию;
- Способы подключения;
- Основные методы и функции;

**4. Установила библиотеку EPPlus  с помощью Package Manager.**

- Ссылка на последнюю версию EPPlus: [https://www.nuget.org/packages/EPPlus/5.0.1-beta](url)

- Написала простейшую программу с использованием данной библиотеки

dimabeel, 2020-03-03:
Внимание, у EPPlus меняется политика лицензирования. 
https://github.com/JanKallman/EPPlus - этот репо все ещё по старой лицензии LGPL
https://github.com/EPPlusSoftware/EPPlus - новый репо с новой лицензией. Его использовать нельзя.

dimabeel, 2020-04-07:
https://www.e-iceblue.com/Introduce/free-xls-component.html

PR #278 (2020-04-15): New excel lib (Free Spire)
Fixes #93.

- [x] Переделка кода
- [x] Заброс файлов в теневое хранилище
- [x] Исправление ошибок во время выполнения
- [x] Настройка форматирования
- [x] Рефакторинг

Будет заменено на Free Spire.XLS
closed 2020-04-15 by idzm