sav-oss-eplanner/EasyEPLANner #96: Отсутствует в Eplaner бачок лед воды

KUZMA-1989, 2020-01-29:
При описании базового объекта отсутствует тип бачок лед воды

dimabeel, 2020-01-29:
@KUZMA-1989 , а какие ещё "бачки" есть? Помимо обычного танка и бачка с ледяной водой?

dimabeel, 2020-01-30:
Я бы предположил использовать префикс перед TANK. Что бы получилось типа type_TANK.
Например:
1. Бачок ледяной воды - CW_TANK (в том числе бачок откачки ледяной воды)
2. Бачок конденсата - C_TANK (В том числе бачок откачки конденсата)
3. Бачок промежуточный - IN_TANK

Остальные бачки на картинке - танки с таким названием. (со слов @idzm).

KUZMA-1989, 2020-01-30:
@dimabeel промежуточный бачок это что?

dimabeel, 2020-01-30:
Не знаю. Было в каком-то из проектов. Если сейчас нигде не используется, можно убрать

KUZMA-1989, 2020-01-30:
@dimabeel  Пусть будет так

dimabeel, 2020-01-30:
@KUZMA-1989 , хорошо, договорились.

PR #98 (2020-01-30): Added small tank to base objects
Fixes #96.

**Решение**
Добавлен отсутствующий объект. Он будет под именем "Бачок".
closed 2020-01-30 by rartem