sav-oss-eplanner/EasyEPLANner #97: Автоматическое выключение программы при бездействии более определенного времени

dimabeel, 2020-01-29:
Необходимо реализовать функционал, который бы позволял безопасно закрывать Eplan если в нем не работают/бездействуют более 1 часа. Необходимо, что бы перед закрытием текущие проекты были закрыты и сохранены, а после этого закрывалась сама программа.
Также следует предусмотреть, что бы по истечении 1 часа бездействия пользователю выводилось окошко, которое бы предупреждало, что через 10 секунд программа будет закрыта (должен вестись отсчет времени). В этом окошке должна быть кнопка "Отменить", которая бы отменяла выключение до следующего прошедшего часа бездействия.

Пример ситуации: конец рабочего дня, персонал закончил работать, пошел домой, а Eplan забыл закрыть. Лицензии заняты. Или человек ушел в отпуск/заболел и тд и тп.

AS005008, 2020-01-31:
Задание получила, вижу. Буду пробовать

AS005008, 2020-02-13:
Если выполняется условие ( бездействие 1 час)
тогда{
Блок 1
запускаем messagebox(с кнопкой отмена) 
запускаем таймер (10с)
Блок 2
Окошко закрывается через 10 секунд
Сохраняются автоматически все изменения в программе
Выход из программы
В обратном случае
Блок 2.1
если пользователь нажал кнопку отмена, тогда
закрываем окошко немедленно и ничего более не делаем

idzm, 2020-02-13:
> Если выполняется условие ( бездействие 1 час)
> тогда{
> Блок 1
> запускаем messagebox(с кнопкой отмена)
> запускаем таймер (10с)
> Блок 2
> Окошко закрывается через 10 секунд
> Сохраняются автоматически все [...]

Рекомендую переоформить алгоритм для более чёткого восприятия. Образец, например, можно глянуть [здесь](https://neerc.ifmo.ru/wiki/index.php?title=%D0%90%D0%BB%D0%B3%D0%BE%D1%80%D0%B8%D1%82%D0%BC_LZW).

AS005008, 2020-02-18:
Шаг 1. Если выполняется условие бездействия, то открываем окошко с предупреждением закрытия. 
Шаг 2. Запускаем таймер с отсчётом 10 секунд.
Шаг 3. Если пользователь не закрыл таймер, тогда сохраняем изменения в программе и выходим из неё.

dimabeel, 2020-02-20:
@AS005008 Перехватываю этот запрос у Вас т.к появилась необходимость в скором добавлении такого функционала в надстройку.

AS005008, 2020-02-20:
Хорошо, можно мне тогда другие задачки поставить( можно мелкие) и по
другому вопросу, чтоб я не бездействовала.

dimabeel, 2020-02-20:
@AS005008 , для начала закройте оставшиеся :)

dimabeel, 2020-04-13:
Не работает с большими числами. Через час не отключается, непонятно почему. На небольших числах (например 5 минут) работает. Скорее всего проблема в расчетах. Надо дорабатывать.

AS005008, 2020-04-14:
Задание приняла. Буду смотреть

dimabeel, 2020-04-14:
> […](#)

Вам не нужно. Я не знаю чего гитхаб отправил письмо.

AS005008, 2020-04-14:
А мне задание вообще будет?

dimabeel, 2020-04-15:
> […](#)

Надо у @idzm спрашивать, он этим занимается

PR #142 (2020-02-24): Creating idle application module
Fixes #97.

Готовый функционал будет расписан после завершения разработки.
PR #300 (2020-04-17): Corrected idle module algorithm
Fixes #97.

Изменен алгоритм работы. Если раньше все время проверялось с эталоном (1 час), то теперь итеративный подход. Каждую минуту (эта цифра может быть изменена) проверяется активность пользователя. Если пользователь не активен более минуты, то произойдет инкрементальное увеличение счетчика на единицу. Таких увеличений будет ровно столько, сколько нужно для того, что бы достичь желаемого времени простоя (1 час = 60 минут = 60 проверок) - это число тоже можно изменять. Таким образом, система будет проверять, если более 60 раз время бездействия будет более 1  [... cut]
closed 2020-04-17 by idzm