sav-oss-eplanner/EasyEPLANner #106: Переименование клапанов мойки танка при копировании танка

dimabeel, 2020-01-31:
При копировании танка, меняется в действии только номер танка. Но т.к в операции мойки танка есть ещё и клапана, номер которых равен номеру танка, то их надо тоже переименовывать.

Но тут есть проблема. Нельзя отследить, какой конкретно клапан будет "принадлежать" танку. Тут возникает архитектурный барьер. Танк ничего не должен знать о линии, а линия ничего не должна знать о танке. Но в нынешней реализации это не разделено. Стоит вопрос о разделении на два объекта: Танк и Линия мойки танка. Но это в перспективе.

Надо прорабатывать этот вопрос с @idzm.

idzm, 2020-07-23:
Да - вопрос решается путём разделения на два объекта: `Танк` и `Линия мойки`.

idzm, 2020-07-23:
Предлагается закрыть данный вопрос.

dimabeel, 2020-07-23:
Хорошо. Будем разделять

closed 2020-07-23 by dimabeel