sav-oss-eplanner/EasyEPLANner #108: Доработка алгоритма копирования объекта

dimabeel, 2020-01-31:
При копировании объекта например с именем FIRST_TANK, в объект SECOND_TANK другого типа, то копируются названия прописанных устройств из источника, потом меняешь название объекта на SECOND_TANK в приемнике, сразу меняются имена **включенных** устройств на новое, а в **выключить** не меняются. Надо доработать, что бы он все поля проверял.

dimabeel, 2020-02-03:
Протестировал. Все работает. Затык лишь в том, что меняются ОУ конкретного танка. Например скопировали Танк 1, то поменяются только ОУ танка 1. Но, например, в операции "Мойка" есть клапана и других танков. Так вот, клапана других танков не будут меняться, и по логике не должны т.к они не принадлежат танку 1.

dimabeel, 2020-02-03:
Со слов @idzm  "Это будет в новой версии 2.0 - когда будет базовый объект общий одинаков. Пока это не будет реализовано." То есть, сейчас этот вопрос можно отложить

idzm, 2020-07-23:
Для решения этой проблемы все устройства, которые не связаны с объектом (танк, линия и т.п.) выделяются в другие - связанные агрегаты. Например, `Линия мойки`.

closed 2020-07-23 by idzm