sav-oss-eplanner/EasyEPLANner #144: Ограничения внутри объектов одинакового типа

KUZMA-1989, 2020-02-21:
Сделать, чтоб в ограничениях внутри объекта с одинаковым типом была синхронизация. Если изменяешь в одном объекте, то в о всех также изменялись ограничения

dimabeel, 2020-02-21:
@KUZMA-1989 , представим ситуацию, когда объект с одинаковым типом, а ограничения должны быть разными. Что делать в таком случае? Система их обновит и ты не сможешь настроить ограничения т.к она постоянно будет их синхронизировать. Или такая ситуация невозможна?

KUZMA-1989, 2020-02-21:
Один тип он и есть один тип, все одинаково

dimabeel, 2020-02-21:
@KUZMA-1989 , хорошо, я понял

dimabeel, 2020-02-24:
@KUZMA-1989 , смотрю проект для примера. Там четыре объекта одинакового типа. Но у них отличаются ограничения. Почему? Это противоречит этому запросу.

Объект 1:
[image]

Объект 2:
[image]

Объект 3:
[image]

Объект 4:
[image]

Тип один и тот же. Почему отличаются ограничения? В частности, последние 3 ограничения на каждом из объектов.

KUZMA-1989, 2020-02-24:
@dimabeel  ограничения внутри объекта поле, там все одинаково 5,2 6,2 7,2 8,2, первый номер это номер объекта, а вторая цифра это операция

KUZMA-1989, 2020-02-24:
Нас интересуют именно "Ограничения внутри объекта"

dimabeel, 2020-02-24:
Я понял

dimabeel, 2020-02-24:
@KUZMA-1989 , а ограничение "На последующие операции" используется? Третий тип ограничений, в самом низу списка ограничений.

KUZMA-1989, 2020-02-24:
@dimabeel Если они относятся к операциям только данного объекта, то да

PR #148 (2020-02-25): Synchronizing restrictions in object
Fixes #144.

**Решение**
Синхронизация ограничений "Внутри объекта" при добавлении или удалении ограничения в одном из объектов.
closed 2020-02-25 by idzm