sav-oss-eplanner/EasyEPLANner #154: Реализовать описание аппарата

idzm, 2020-02-26:
Для аппарата (это в основном танки) реализовать следующие ниже элементы описания.
- [ ] оборудование: 
```lua
    --Мойка
    self.hatch	= GS( 'TANK'..self.n..'GS1' )		--Датчик люка.

    --Набор
    self.LS_up	= LS( 'TANK'..self.n..'LS2' )		--Датчик верхнего уровня.
    self.LS_down= LS( 'TANK'..self.n..'LS1' )		--Датчик нижнего уровня.
    self.LT	= LT( 'TANK'..self.n..'LT1' )		--Датчик текущего уровня.

    self.temperature = TE( 'TANK'..self.n..'TE1' )	--Датчик температуры.
```
- [ ] операция наполнения:
```lua
self.operations.FILL = 2				--Номер операции наполнения.
self.OPERATION_AFTER_FILL = self.operations.STORING	--Операция после набора.
```

- [ ] операция хранения:
```lua
self.operations.STORING = 3
```

- [ ] операция выдачи:
```lua
self.operations.OUT = 4
```

- [ ] параметры, сбрасываемые после начала мойки:
```lua
self.reset_before_wash =
    {
    self.PAR_FLOAT.V_ACCEPTING_CURRENT,
    self.PAR_FLOAT.PRODUCT_TYPE,
    self.PAR_FLOAT.V_ACCEPTING_SET,
    }
```

- [ ] мастер проекта:
```lua
self.master = prg.master1
```

dimabeel, 2020-02-26:
@idzm 
> self.OPERATION_AFTER_FILL = self.operations.STORING	--Операция после набора.

Каким образом задавать это?

dimabeel, 2020-02-26:
- [ ] модификация доп. сигналов моек:
```lua
prg.tank10.DI_CIP_FREE = prg.control_modules.MCA2LINE2DI1
```

dimabeel, 2020-02-26:
Также:

- [ ] Сделать, что бы можно выбрать только 1 конкретную базовую операцию на объект

PR #153 (2020-03-02): Improving base tank functionality
Fixes #154.
closed 2020-03-02 by idzm