sav-oss-eplanner/EasyEPLANner #180: Добавить датчики

dimabeel, 2020-03-11:
Сравнить, какие датчики из этих есть в надстройке, добавить недостающие. Проработать варианты (совместно с @id и @rartem), что бы систематизировать датчики с одинаковыми типами, подтипами но с отличающимися областями ввода/вывод IO-Link.

Производитель | Каталожный   номер | Название | Описание
-- | -- | -- | --
**Phoenix** | 2910411 | CBMC E4 24DC/1-10A   IOL | Электронный защитный   выключатель (#61)
**Festo** | 8001446 | SPAE-P10R-S6-PNLK-2.5K | датчик давления 0-10   bar IO-Link
**Festo** | 573384 | VAEM-L1-S-8-PT | Модуль IO-Link
**Festo** | 573939 | VAEM-L1-S-16-PT | Модуль IO-Link
**Festo** | 573940 | VAEM-L1-S-24-PT | Модуль IO-Link
**Endress+Hauser** | FTL33-GR7N2ABW5J | Датчик предельного   уровня | G3/4",М12,   IO-Link, PNP
**ALFA LAVAL** | 9615400306 | Think Top V70 | IO-link 3*3/2 M12
**ALFA LAVAL** | 9615400308 | Think Top V70 | IO-link 1*5/2 M12
**BURKERT** | 8694-E1-00-FA03-T-N-MPGW-0-0-0-Q | Блок управления   клапаном 8694 IO-Link | для привода ELEMENT
**IFM** | LMT100 | Датчик предельного   уровня | М12, G ½ A, 1 NO+1NC   PNP, 18...30 DC, IP 69K
**IFM** | LDL100 | Датчик проводимости | М12, G ½ A,4…20mA, IP   69K, IO-Link 1.1
**IFM** | LDL200 | Датчик проводимости | М12, G 1 A, 4…20mA,   IP 69K, IO-Link 1.1
**IFM** | TA2502 | Датчик температуры | 30 мм, М12, G ½ A,   4…20mA, IP 69K, IO-Link 1.1
**IFM** | TA2435 | Датчик температуры | 100 мм, М12, G ½,   4…20mA, IP 69K, IO-Link 1.1
**IFM** | PM1704 | Датчик давления | G 1 A, -1…10Bar, M12,   4…20mA, IO-Link 1.1
**IFM** | PM1715 | Датчик давления | G 1 A, -1…6Bar, M12,   4…20mA, IO-Link 1.1
**IFM** | PM1705 | Датчик давления | G 1 A, -1…4Bar, M12,   4…20mA, IO-Link 1.1
**IFM** | PM1709 | Датчик давления | G 1 A, -1…1Bar, M12,   4…20mA, IO-Link 1.1
**IFM** | PM1708 | Датчик давления | G 1 A,   -0,0125…0,25Bar, M12, 4…20mA, IO-Link 1.1
**IFM** | PI2715 | Датчик давления | G 1 A, -1...6Bar,   M12, 4…20mA, IO-Link, 2DO
**IFM** | PI2794 | Датчик давления | G 1 A, -1…10Bar, M12,   4…20mA, IO-Link, 2DO
**IFM** | LR2750 | Датчик текущего   уровня (радар) |  
**IFM** | TP3231 | Преобразователь   IO-Link - PT | Pt100/Pt1000-IO-Link/4-20mA
**IFM** | DP1213 | Преобразователь   IO-Link - 2AO | IO-Link - 2AO(4-20mA)
**IFM** | DP2200 | Преобразователь   IO-Link - AI | IO-Link - AI(4-20mA)

По каждому датчику будет отдельное сообщение ниже.

dimabeel, 2020-03-11:
Phoenix | 2910411 | CBMC E4 24DC/1-10A IOL | Электронный защитный выключатель (#61)
-- | -- | -- | --

Рассматривается в #61. Продублируется информация тут.

Область выходных данных:
[image]
Биты почему-то задом наперед идут.

Область входных данных:
[image]

dimabeel, 2020-03-11:
Festo | 8001446 | SPAE-P10R-S6-PNLK-2.5K | датчик давления 0-10 bar IO-Link
-- | -- | -- | --

Датчик уже добавлен в #60 

Информация для справки: размер области - 2 байта.
[image]

[image]

В контроллер этот датчик ещё не добавлен. Нужно будет корректировать потом экспортируемые параметры для этого датчика.

dimabeel, 2020-03-11:
Festo | 573384 | VAEM-L1-S-8-PT | Модуль IO-Link
-- | -- | -- | --
Festo | 573939 | VAEM-L1-S-16-PT | Модуль IO-Link
Festo | 573940 | VAEM-L1-S-24-PT | Модуль IO-Link

Эти датчики уже добавлены: устройство **Y**, подтипы: _DEV_VTUG8_, _DEV_VTUG16_, _DEV_VTUG24_

dimabeel, 2020-03-11:
Endress+Hauser | FTL33-GR7N2ABW5J | Датчик предельного уровня | G3/4",М12, IO-Link, PNP
-- | -- | -- | --

[image]

В системе уже есть устройство **LS** с подтипами _LS_IOLINK_MAX_, _LS_IOLINK_MIN_ от компании IFM, этот - от Endress Hauser. Надо думать, как совместить.

dimabeel, 2020-03-11:
ALFA LAVAL | 9615400306 | Think Top V70 | IO-link 3*3/2 M12
-- | -- | -- | --
ALFA LAVAL | 9615400308 | Think Top V70 | IO-link 1*5/2 M12

**Уже добавлены.** Устройство - **V**, подтипы: *V_IOLINK_MIXPROOF*, *V_IOLINK_DO1_DI2*

dimabeel, 2020-03-11:
IFM | LMT100 | Датчик предельного уровня | М12, G ½ A, 1 NO+1NC PNP, 18...30 DC, IP 69K
-- | -- | -- | --

**Уже добавлен**. Устройство - **LS**, подтипы: _LS_IOLINK_MIN_, _LS_IOLINK_MAX_. Этот датчик конфликтует с E&H Luiquiphant FTL33.

Process data:
[image]

dimabeel, 2020-03-11:
IFM | LDL100 | Датчик проводимости | М12, G ½ A,4…20mA, IP 69K, IO-Link 1.1
-- | -- | -- | --
IFM | LDL200 | Датчик проводимости | М12, G 1 A, 4…20mA, IP 69K, IO-Link 1.1

Process data **LDL100**:
[image]

Proces data **LDL200**:
[image]

Отличаются лишь максимальными значениями, data sheet одинаковый. Нет необходимости как-то совмещать эти датчики, они будут работать.
**Уже добавлен** - **QT**, подтип **QT_IOLINK**

dimabeel, 2020-03-11:
IFM | TA2502 | Датчик температуры | 30 мм, М12, G ½ A, 4…20mA, IP 69K, IO-Link 1.1
-- | -- | -- | --
IFM | TA2435 | Датчик температуры | 100 мм, М12, G ½, 4…20mA, IP 69K, IO-Link 1.1

Process data **TA2502**:
[image]

Process data **TA2435**:
[image]

Datasheet не отличается. Конфликтов с другими датчиками нет
**Уже добавлены**. Устройство - **TE**, подтип: _TE_IOLINK_.

dimabeel, 2020-03-11:
BURKERT | 8694-E1-00-FA03-T-N-MPGW-0-0-0-Q | Блок управления клапаном 8694 IO-Link | для привода ELEMENT
-- | -- | -- | --

Это устройство работает от стандартного сигнала 4..20мА или через AS-интерфейс. Не знаю, как оно попало в список.
Ссылка на устройство - https://www.burkert.com.ru/ru/type/8694
Ссылка на мануал - https://www.burkert.com/en/Media/plm/MAN/MA/MA8694-Manual-EU-EN.pdf?id=MAN0000000000000001000329139ENC

Там нет никакой Process data. Тип устройства для EasyEplanner мне непонятен.

dimabeel, 2020-03-11:
1|IFM | PM1704 | Датчик давления | G 1 A, -1…10Bar, M12, 4…20mA, IO-Link 1.1
--|-- | -- | -- | --
2|IFM | PM1715 | Датчик давления | G 1 A, -1…6Bar, M12, 4…20mA, IO-Link 1.1
3|IFM | PM1705 | Датчик давления | G 1 A, -1…4Bar, M12, 4…20mA, IO-Link 1.1
4|IFM | PM1709 | Датчик давления | G 1 A, -1…1Bar, M12, 4…20mA, IO-Link 1.1
5|IFM | PM1708 | Датчик давления | G 1 A, -0,0125…0,25Bar, M12, 4…20mA, IO-Link 1.1
6|IFM | PI2715 | Датчик давления | G 1 A, -1...6Bar, M12, 4…20mA, IO-Link, 2DO
7|IFM | PI2794 | Датчик давления | G 1 A, -1…10Bar, M12, 4…20mA, IO-Link, 2DO

Эти датчики будут иметь конфликты. Устройство - **PT**. Подтип - _PT_IOLINK_. Добавлены датчики с номерами **6**,**7**. 
Датчики **1**, **2**, **3**, **4**, **5** отличаются от **6**, **7**.

Внимание - отличается _Gradient_ и _Value range_.
Process data: 
1. **PM1704**
[image]

2. **PM1715**
[image]

3. **PM1705**
[image]

4. **PM1709**
[image]

5. **PM1708**
[image]

6. **PI2715**
[image]

7. **PI2794**
[image]

dimabeel, 2020-03-11:
IFM | LR2750 | Датчик текущего уровня (радар) |  -
-- | -- | -- | --

Конфликт. Устройство - **LT**, подтип _LT_IOLINK_. Конфликт на почве того, что для этих же целей используется **IFM PM1705** (номер 3 в сообщении).

Process data:
1. **LR2750**
[image]

2. **IFM PM1705**
[image]

dimabeel, 2020-03-11:
IFM | TP3231 | Преобразователь IO-Link - PT | Pt100/Pt1000-IO-Link/4-20mA
-- | -- | -- | --
IFM | DP1213 | Преобразователь IO-Link - 2AO | IO-Link - 2AO(4-20mA)
IFM | DP2200 | Преобразователь IO-Link - AI | IO-Link - AI(4-20mA)

Какие-то преобразователи. Таких устройств в надстройке нет, и куда их добавлять - неизвестно. Описание устройства в последнем столбце таблицы устройства. Описывается, что они делают.

Process data:
1. **TP3231** https://www.ifm.com/ru/ru/product/TP3231?tab=documents
[image]

2. **DP1213** https://www.ifm.com/ru/ru/product/DP1213?tab=documents

Входные данные:
[image]

Выходные данные:
[image]

3. **DP2200**
[image]

dimabeel, 2020-03-12:
Endress&Hauser | TM311-AAC0BH2BBB2A1| Датчик температуры | -
 -- | -- | -- | --

Добавляется следующий **TE** датчик типа _TE_IOLINK_
[image]

dimabeel, 2020-04-16:
Идея:
1. Сохраняется изделие в файле (для управляющей программы)
2. Переходим к идентификации устройств по изделиям (для расчета иолинка в надстройке). Список изделий в lua файл выносится, а также там указывается тип и подтип датчика, который потом автоматически устанавливается при синхронизации (если установлено изделие). Этот вопрос будет проработан позже.

closed 2020-04-17 by dimabeel

asutp-user, 2020-06-19:
Проверить единицы измерения и gradient на датчике PM1709 (а лучше на всех PM)