sav-oss-eplanner/EasyEPLANner #235: Возможность задания использования технологических агрегатов

idzm, 2020-03-24:
Если для танка добавлен узел охлаждения, то тогда необходимо добавить возможность использования его во время операций наполнения, хранения и выдачи. Сохранение соответственно выглядит так:
```lua
prg.tank1.NEED_COOLING_DURING_FILL = true
prg.tank1.NEED_COOLING_DURING_STORING = true
prg.tank1.NEED_COOLING_DURING_OUT = true
```
или так:
```lua
prg.tank1.NEED_COOLING_DURING_FILL = false
prg.tank1.NEED_COOLING_DURING_STORING = false
prg.tank1.NEED_COOLING_DURING_OUT = false
```

dimabeel, 2020-03-31:
@idzm, предлагаю писать вот так:

```lua
prg.tank1.FILL.NEED_COOLING = true
prg.tank1.STORING.NEED_COOLING = true
prg.tank1.OUT.NEED_COOLING = true
```
или так:
```lua
prg.tank1.FILL.NEED_COOLING = false
prg.tank1.STORING.NEED_COOLING = false
prg.tank1.OUT.NEED_COOLING = false
```

Имя свойства `NEED_COOLING` обсуждаемо

idzm, 2020-04-01:
Да, пойдёт - `NEED_COOLING` может быть.

PR #261 (2020-04-02): Added ability to use aggregates in apparats
Fixes #235.

- [x] Добавление параметров для танка, при вводе агрегата (узла охлаждения)
- [x] Удаление параметров из танка при удалении из привязанных агрегатов (узла охлаждения)
- [x] Сохранение параметров агрегатов танка.
- [x] Чтение параметров агрегатов танка из lua.

Список может дополняться.

- [x] Рефакторинг
- [x] Ошибка при сохранении, если указаны несуществующие объекты (т.к Save раньше, чем Check). Ошибка, если объект нельзя привязать (после сдвигов)
- [x] Тестирование
- [x] Документация
closed 2020-04-02 by idzm