sav-oss-eplanner/EasyEPLANner #270: Доработка базовых агрегатов с ПИД-регулированием

idzm, 2020-04-03:
Для всех текущих  базовых агрегатов с ПИД-регулированием добавляется дополнительное оборудование (*AI1/номер параметра*) для указания задания ПИД-регулятора:
```lua
prg.flow_node_pid_o2.SET_VALUE_PAR_NUMBER_OR_AI1 = 2
--или 
prg.flow_node_pid_o2.SET_VALUE_PAR_NUMBER_OR_AI1 = prg.control_modules.O2AI1
```
Оно не может быть пустым.

KUZMA-1989, 2020-04-03:
@idzm А то, что я писал сделать для ПИД вход и задание поля универсальные не подходит?

idzm, 2020-04-03:
Это и есть - здесь я конкретно название указал и как в Eplanner будет называться.

idzm, 2020-04-03:
Также предлагается вместо номера параметра использовать Lua-имя:
```lua
prg.flow_node_pid_o2.SET_VALUE_PAR_NUMBER_OR_AI1 = prg.flow_node_pid_o2.PAR_FLOAT.HEAT_TEMPERATURE
```

PR #271 (2020-04-03): Added equipment for PIDs and its checking
Fixes #270.
closed 2020-04-03 by idzm