sav-oss-eplanner/EasyEPLANner #281: Автоматическое включение мойки по сигналам

KUZMA-1989, 2020-04-08:
Сделать поле "Автоматическое включение мойки", чтобы прописывать туда сигналы.

Логика следующая: 

- Если в поле "Автоматическое включение мойки" есть сигнал, а в поле "Завершение мойки" нет сигнала, то мойка включается по сигналу из поля "Автоматическое включение мойки" и выключается по его пропаданию.

- Если в поле "Автоматическое включение мойки" есть сигнал и в поле "Завершение мойки" есть сигнал, то мойка включается по сигналу из поля "Автоматическое включение мойки", а по сигналу из поля "Завершение мойки" либо выключается, либо переходит в шаг дренаж, если таковой имеется.

dimabeel, 2020-04-08:
@KUZMA-1989 с логикой к @idzm. Куда нужно добавить это поле? В какие объекты? В какие операции?

KUZMA-1989, 2020-04-08:
@dimabeel Для всех объектов

dimabeel, 2020-04-08:
@KUZMA-1989 , то есть, по этому сигналу будет включаться мойка какого-то конкретного объекта, верно?

KUZMA-1989, 2020-04-08:
@dimabeel Да, это поле в объекте, можно добавить рядом с полем мойка завершена

dimabeel, 2020-04-08:
@KUZMA-1989 , мойка завершена - это сигнал в операции "Мойка" для объектов, это не сигнал конкретного объекта. Я думаю, что этот сигнал стоит добавить в операцию "Мойка"

KUZMA-1989, 2020-04-08:
Да, именно так, у меня eplan не открыт просто

dimabeel, 2020-04-08:
@KUZMA-1989 , хорошо. Я понял

idzm, 2020-04-08:
Сигнал "Автоматическое включение мойки" для сохранения описания будет называться ```CIP_WASH_REQUEST```.

PR #283 (2020-04-08): Added wash request parameter
Fixes #281.
closed 2020-04-08 by idzm