sav-oss-eplanner/EasyEPLANner #350: Добавить базовый агрегат "Бойлер"

idzm, 2020-05-14:
Lua-имя `basic_boiler`. Операции и шаги:
```lua
boiler =
    {
    operations =                            --Операции.
        {
        HEATING                  = 1,       --Нагрев.
        },

    steps =
        {
        HEATING =
            {
            WAITING_LOW_LS = 1,
            WATER_2_LOW_LS = 2,
            WATER_2_HI_LS = 3,
            }
        }
    }
```

dimabeel, 2020-05-14:
@idzm , что такое WATER_2?

idzm, 2020-05-14:
Возможность использования - да, нужна - свойство `boiler`.

dimabeel, 2020-05-14:
@idzm , а WATER2, это что?

idzm, 2020-05-14:
Шаги - ожидание пропадания нижнего уровня, наполнение до нижнего уровня, наполнение до верхнего уровня соответсвенно.

PR #357 (2020-05-14): Implemented base Boiler object
Fixes #350

- [x] Уточнить свойство и название шагов
- [x] Протестировать
- [x] Документация

Пример свойства в операции, для использования узла во время выполнения:
```LUA
prg.tank3.STORING =
    {
    BOILER = false,
    }
```
closed 2020-05-14 by idzm