sav-oss-eplanner/EasyEPLANner #373: Оптимальное сохранение параметров

idzm, 2020-05-26:
Предлагается при сохранении описания (в **prg.lua**, **main.objects.lua**) для полей, у которых нет значения (значение **nil**), опускать их запись. Это позволить более оптимально (наглядно) получить описание - если поле не задано, значит его и нет в описании.

dimabeel, 2020-05-26:
Хорошо, сделаю после того, как разберусь с экспортом/импортом

dimabeel, 2020-05-28:
@idzm , для steps, operations в prg.lua делаем аналогично? Пустую таблицу можно убрать, или лучше все же оставить?

dimabeel, 2020-05-28:
Если параметр не сохраняется, и в этой таблице параметров больше нет, то пустую таблицу я тоже не пишу, например:

До:

prg.tank1.FILL =
```LUA
{
OPERATION_AFTER_FILL = nil,
}
```

После: этот код написан не будет.

idzm, 2020-05-28:
Да - пустые таблицы тоже не пишем.

PR #375 (2020-05-28): Not saving empty parameters and equipment
Fixes #373.

Для параметров и оборудования введены проверки, которые позволяют не сохранять пустые параметры и оборудование в файл т.к их нету, то и в файле их быть не должно.
closed 2020-05-28 by idzm