sav-oss-eplanner/EasyEPLANner #380: Доработка шаблона базового описания проекта в main.plua

idzm, 2020-05-28:
Шаблон вынести в отдельный файл `main.plua`.
Вот его содержание:
```lua
--Проект 'Название_проекта'

prg = require("prg")                           --Технологические объекты.
------------------------------------------------------------------------------

require( "sharedmodbus" )               -- Межконтроллерный обмен
require( "shared" )
------------------------------------------------------------------------------
------------------------------------------------------------------------------

--Пользовательская функция инициализации, выполняемая однократно в PAC.
function user_init()
end
------------------------------------------------------------------------------
------------------------------------------------------------------------------

--Пользовательская функция, выполняемая каждый цикл в PAC.
function user_eval()
end
------------------------------------------------------------------------------
------------------------------------------------------------------------------

--Функция инициализации параметров, выполняемая однократно в PAC.
function user_init_params()
end
------------------------------------------------------------------------------
------------------------------------------------------------------------------

```

dimabeel, 2020-05-28:
#342 можно удалить?

idzm, 2020-05-28:
Оставляй, просто в решении закроешь сразу два вопроса.

dimabeel, 2020-05-28:
@idzm , что значит "Вынести в отдельный файл?" Сейчас шаблон из ресурсов берется проекта, зачем выносить его в файл? Этот файл очень мало меняется

idzm, 2020-05-28:
Планируется его серьезное изменение, поэтому готовимся заранее к этому.

dimabeel, 2020-05-28:
@idzm , предлагаю тогда по аналогии с .lua скриптами, файл будет в проекте и шаблон будет браться оттуда

idzm, 2020-05-29:
Да - так и делаем.

dimabeel, 2020-05-29:
@idzm , сейчас файл в Windows1251 кодировке. Переходим на UTF-8 или оставляем прежнюю?

idzm, 2020-05-29:
Пока оставляем 1251.

PR #381 (2020-06-01): Updated main.plua pattern
Fixes #380.
Fixes #342.
closed 2020-06-01 by idzm