sav-oss-eplanner/EasyEPLANner #400: Указание операции перемешивания для узла при наличии более 1-ой таковой

idzm, 2020-06-02:
При наличии реверса у узла перемешивания появляются две операции: перемешивание влево и перемешивания вправо (мотор мешалки работает в реверсе).
Во всех операциях должен работать узел перемешивания: в операциях мойка, наполнение и перемешивания - влево, в операциях внесения закваски, отделение сыворотки, хранение, выдача - вправо. У инженеров по автоматизации должна быть возможность это указывать (и изменять при необходимости).

dimabeel, 2020-06-03:
Посмотрев #401, есть несколько путей развития:
1. Я могу добавить просто привязочные параметры в каждую операцию при привязке узла перемешивания (параметр - номер операции перемешивания в узле перемешивания). Это самый простой способ, он требует только введения обработки такого типа параметра в операции "Мойка";
2. Подумать над переработкой параметров. Появятся отдельные параметры операций, отдельные параметры оборудования (разные классы). Параметры операций также будут разделены на те, в которых указываются: устройства и параметры, номера операций, прочие данные, булевые значения (пока что четыре типа, может будет больше).
3. По принципу первого параметра, только ещё будет дописан метод обработки обработки номеров введенных операций, что бы операция записывалась в prg.lua её адресом (Prg.mix.operations.mixing), а не цифрой.

@idzm , что скажешь?

dimabeel, 2020-06-03:
Я хочу попробовать оставить параметры в нынешнем положении, но ввести тригеры введенных значений: isBool (уже есть), isDevice (ещё нет), isParameter (ещё нет), isNumber (ещё нет). В итоге можно будет обработать в зависимости от введенного значения параметр. Если булевское (для булевского параметра, он остается классом) - одно сохранение, если введено устройство - другое и т.д. В перспективе мы можем легко расширять это поведение т.к у нас всегда одно сохранение:

1. Для устройства - prg.control_modules.DEVICE
2. Для параметра - object.PAR_FLOA.PARAMETER
3. Для числа - сейчас числами мы указываем только номера операций (если будет что-то ещё, то придется переделывать по другому, или такой вариант всегда будет?)
4. Для булевского - true/false

idzm, 2020-06-03:
Пойдет - с типами и проверки введенных значений проще реализовывать.

dimabeel, 2020-06-03:
@idzm , что если, перемешивание в операции не должно выполняться. Что тогда?

idzm, 2020-06-03:
Надо добавить параметр - использовать перемешивание во время операции.

idzm, 2020-06-03:
Если перемешивание не нужно, то дальше можно по разному: или не отображать, или 0 писать.

dimabeel, 2020-06-03:
> Если перемешивание не нужно, то дальше можно по разному: или не отображать, или 0 писать.

Это слишком уникальный случай на текущий момент. Нужно думать, что делать. Тут быстрыми решениями точно не отделаться

dimabeel, 2020-06-03:
Также, приведение к одному типу сохранения привело к тому, что мы должны отказываться от задания параметров непосредственно в самом объекте:

```LUA
prg.OBJECT1.CIP_FREE = <value>
```

должно перейти в:

```LUA
prg.OBJECT1.WASHING_CIP = {
     CIP_FREE = <value>
}
```

Это коснется всех параметров операции "Мойка", которые реализованы на текущий момент:

```
CIP_WASH_REQUEST
CIP_WASH_END
DI_CIP_FREE
MEDIUM_CHANGE_REQUEST
```

idzm, 2020-06-03:
Пойдет. Только упрощаем:
```LUA
prg.OBJECT1.WASH = 
    {
    CIP_FREE = <value>
    }
```

dimabeel, 2020-06-03:
А что будет с файлами main.plua? Там же используется prg.OBJECT1.CIP_WASH

idzm, 2020-06-03:
Да.

dimabeel, 2020-06-03:
В данный момент реализовано всё, что запрошено в issue, но нужно подумать по поводу того, что бы отключить/удалить/сделать неактивными/обнулить параметры, которые добавляются агрегатами если в операции они не будут использоваться. Поэтому, если нужный issue нужен сейчас, то я предлагаю вынести это в отдельный запрос. А если не срочно - то тут надо работать дальше (включая тестирование, документацию)

idzm, 2020-06-03:
Пока можно оставить параметры как есть - в неиспользуемые пусть 0 пишется.

dimabeel, 2020-06-03:
> Пока можно оставить параметры как есть - в неиспользуемые пусть 0 пишется.

Это тоже пока не реализуемо. Нужно следить за конкретным параметром. Любое изменение параметра использования узла "Да/Нет" никак не отследить на данный момент. Тут или дорабатывать разом, или оставлять пока так, как есть - ставишь "Нет" и очищаешь поле. Единственное, я ещё добавлю, что бы поле можно было очистить по Del.

idzm, 2020-06-03:
Ок - оставляем пока такой вариант.

PR #403 (2020-06-09): Selecting mixing type in object
Fixes #400.

- [x] Проработать вопрос
- [x] Переделать параметры объекта как агрегата, чтобы он мог принимать активный параметр
- [x] Сохранение в зависимости от введенного типа значения в поле (bool, device, parameter, number)

**UPD** Вскрылись новые подробности
- [x] ~Подумать над отключением (скрытием) параметров, если отключается использование узла, который добавляет параметры в операции (узел перемешивания, прим.)~
- [x] ~Если это можно реализовать сейчас - реализовать, если нет - на будущие доработки.~
**Будет вынесено в отдельный Issue**

- [x] Тестирова [... cut]
closed 2020-06-09 by idzm