sav-oss-eplanner/EasyEPLANner #443: Получение справки с платформы ostis.savushkin.by

idzm, 2020-06-23:
Необходимо предоставить пользователю получать интеллектуальную справку по основным элементам системы в контексте **S88**. Для этого предлагается по нажатии пользователем **F1** запускать браузер с адресом вида `http://ostis.savushkin.by/?sys_id=recipe`, где `recipe` - ключевое слово, ассоциированное с текущим активным узлом дерева представления описания проекта.

dimabeel, 2020-06-23:
@idzm, это срочно?

idzm, 2020-06-23:
К пятнице желательно сделать - делаем статью в Springer, поэтому желательно вставить реальные скриншоты.

dimabeel, 2020-06-24:
@idzm , какой список ключевых слов должен быть вместо `recipe` и для каких объектов дерева? У меня нет доступа на ostis, я не могу залогиниться под своими учетными данными.

idzm, 2020-06-24:
Пользователь будет самостоятельно вводить свои учетные данные.

PR #445 (2020-06-24): Open tab from browser after press F1 in objects editor
Fixes #443.
closed 2020-06-24 by idzm