sav-oss-eplanner/EasyEPLANner #449: Проверка на корректность описания

idzm, 2020-06-26:
При проверке проекта также добавить на проверку на корректность запуска управляющей программы. Для этого запускать консольный скрипт: 
```
chcp 1251
..\..\spec\main.lua  -o TAP полный_путь\main.plua
```
В случае отсутствия ошибок он возвращает 0, иначе - есть ошибки. При их наличии выводить результаты выполнения в окно сообщений пользователю.
Также должна быть возможность настройки данной функциональности пользователем (возможность изменить скрипт, отключить и т.п.).

dimabeel, 2020-07-09:
@idzm , относительный первый путь вряд-ли подойдет, неизвестно где будет лежать скрипт на рабочей станции инженера.  Он может его положить в другое место. Поэтому путь надо, скорее всего, задавать в документации.

По самому скрипту: я его выполнил и не увидел результатов работы.
[image]
Если это результат, то он непонятен. Как его интерпретировать?

idzm, 2020-07-09:
Скрипт надо локально запускать.

dimabeel, 2020-07-09:
@idzm , скрипт будет запускаться с помощью `Process.Start(application, args)` и в него уже будет передаваться строка. Если запускается от имени администратора **cmd**, то она сразу находится в` c:/Windows/System32`, а если локально, то она находится в папке пользователя `C:/users/username`. 

https://www.cyberforum.ru/csharp-beginners/thread1611076.html - это решение использовать результат выполнение скрипта без промежуточного файла.

dimabeel, 2020-07-20:
@idzm, пробовал запустить скрипт через lua.exe. Что-то не то. Не понимаю, как эти пользоваться

[image]

Видимо через lua.exe я работаю с экземпляром lua, и он консольный скрипт не будет выполнять.

dimabeel, 2020-07-20:
@idzm , развернул получившуюся модель на более понятный вид.
[image]

Нужно разобраться с алгоритмом работы базовой операции, которая будет главной над всеми операциями конкретных агрегатов/аппаратов. Не совсем понимаю, как оно должно работать

dimabeel, 2020-07-20:
`lua ..\..\spec\main.lua  -o TAP -C -путь` - корректный скрипт

PR #495 (2020-07-22): Checking project through CMD
Fixes #449.

`chcp 1251`
`lua ..\..\spec\main.lua  -o TAP -C -полный путь` - корректный скрипт

- [x] Проверка доступности LUA через CMD
- [x] Проверка проекта на корректность через CMD
- [x] Выдача корректных сообщений об ошибках
- [x] Загрузка скрипта из файла
- [x] Возможность отключения, изменения скрипта
- [x] Вынесение отключения и изменения в настройки в .ini
- [x] Рефакторинг
- [x] Тесты
- [x] Документация

Пока что не делал графический интерфейс для настроек. Это будет следующая задача, пока что в .ini
closed 2020-07-22 by rartem