sav-oss-eplanner/EasyEPLANner #455: Добавление виртуальных устройств в список устройств

KUZMA-1989, 2020-07-02:
В устройствах сделать отдельную вкладку VIRT и туда добавить вкладки DI_VIRT, DO_VIRT, AI_VIRT, AO_VIRT, чтоб можно было видеть список сигналов

dimabeel, 2020-07-09:
@KUZMA-1989 , оставлять виртуальные сигналы в группах, которые соответствуют этим сигналам? DI в DI, AI в AI и др.
Например из группы DI убираем VIRT сигналы, и добавляем их в группу DI для VIRT группы? Или надо, что бы оставались в обоих группах?
То есть, получим дублирование сигналов в двух группах.

KUZMA-1989, 2020-07-09:
@dimabeel Сейчас вообще нет в списке сигналов виртуальных
[image]

KUZMA-1989, 2020-07-09:
Сделай отдельные вкладки на 1-м уровне, DI_VIRT, DO_VIRT, AI_VIRT, AO_VIRT

KUZMA-1989, 2020-07-09:
И туда их закинь

KUZMA-1989, 2020-07-09:
Когда нажата кнопка скрывать привязанные, то прячь их

PR #486 (2020-07-17): Added virtual devices to devices window
Fixes #455.
closed 2020-07-17 by rartem