sav-oss-eplanner/EasyEPLANner #499: Отображение в дереве объектов только активных (заполненных) элементов

idzm, 2020-07-23:
Необходимо реализовать возможность отображения в дереве только заполненных элементов. Должен быть элемент (кнопка) на панели инструментов, включающий\отключающий данную функциональность. Это необходимо для разгрузки дерева - меньше элементов, проще навигация.

dimabeel, 2020-07-23:
@idzm, что мы понимаем под словом "элемент"?

idzm, 2020-07-23:
Элемент - узел дерева, заполненный - значит у него отличное от пустого (пустая строка) значение.

dimabeel, 2020-07-23:
ObjectListView не поддерживает скрытие элементов дерева. Можно только заменить отображаемую модель. Надо думать, что можно придумать, потому что содержать две модели не хочется.

dimabeel, 2020-07-24:
**Задача**: скрыть пустые элементы дерева.
**Возможное решение**: Делаем копию дерева, удаляем пустые объекты. Копия делается легко через Clone.
**Проблемы**:
В - Что делать, если меняем "Базовый объект", "Базовую операцию" или любой другой элемент, который приводит к изменению количества объектов в дереве?
О - Дерево должно обработать изменения и отобразить их, если они не пустые.
В - Как это сделать?
О - Неизвестно

В - Как проследить связь между двумя деревьями, т.к через Clone связи между деревом полным и неполным нет?
О - Неизвестно

В - Как восстановить полное дерево?
О - Неизвестно

PR #505 (2020-07-27): Button for show only filled items in tree
Fixes #499.
closed 2020-07-27 by idzm