sav-oss-eplanner/EasyEPLANner #507: Возможность указания для оборудования более чем одного устройства

idzm, 2020-07-28:
В настоящее время устройства указываются так:
```
prg.sterile_air_node_tank1.TE1 = prg.control_modules.TANK1TE1
```
Реализовать возможность задания группы устройств:
```
prg.sterile_air_node_tank2.TE1 = { prg.control_modules.TANK1TE1, prg.control_modules.TANK1TE2 }
```
Пользователь через пробел будет указывать группу устройств. Если устройств одно, то оно должно сохраняться по  старому механизму.

dimabeel, 2020-07-28:
@idzm , почему мы пишем в TE1 ещё и TE2? Это неправильно.

Мы должны изменить LUA имя оборудования. Либо для группы указывать одно имя, а для одиночного типа - другое имя. Автопривязка для множественных вариантов работать не будет (мы не можем указать в defaultValue сразу несколько значений, это неправильно с точки зрения объектов), только для одиночного.

dimabeel, 2020-07-28:
По автопривязке думаю сделать так: привязывать по дефолтному значению, а если пользователь добавил сам, то ничего не делать, потому что датчиков всегда может быть разное количество

idzm, 2020-07-28:
Ок. Пойдёт.

PR #509 (2020-07-29): Read more than one device in equipment
Fixes #507.

- [x] Возможность указания более **1** устройства в оборудовании
- [x] Отключение ошибки об этом
- [x] ~Корректировка привязки (_привязываем **1** устройство, а если добавлено более **1**, то не трогаем_)~ Не понадобилась
- [x] Автоматическая модификация устройств (_когда их более **1** в оборудовании_) при изменении ОУ или номера объекта.
- [x] Рефакторинг
- [x] Тестирование
- [x] Документация
closed 2020-07-29 by idzm