sav-oss-eplanner/EasyEPLANner #514: Добавить клапана BURKERT 8694

dimabeel, 2020-07-30:
Документация - [тык](https://www.burkert.com/en/Media/plm/MAN/MA/MA8694-ManualREV.2-EU-EN.pdf?id=MAN0000000000000001000396587ENB).
IODD файл, описание Process Data - [тык](https://www.burkert.com/en/Media/plm/SOW/SOW/8694_DeviceDescription_A03.zip?id=SOW0000000000000001000396184MLD)
Сайт изготовителя - [тык](https://www.burkert.com.ru/ru/type/8694)

Устройства - **BURKERT Type 8694 REV.2** и **BURKERT Type 8694 IO-Link B** (_Видимо одинаковые с точки зрения ПО_). Самая полная документация только на **REV.2**, работать могут в режиме **AS-i** и **IO-Link** (_два подтипа_)

dimabeel, 2020-07-30:
Из нужного (_как для **IO-Link**, так и для **AS-i**_):
1. Какие параметры будут экспортироваться в базу каналов?
2. Какие параметры будут у устройства?
3. Как будут называться изделия?

dimabeel, 2020-09-24:
Input - 10 byte, output - 5 byte.

idzm, 2020-10-05:
Вот описание IO-Link - [MA8694-8791-8696-IODD-file-EU-ML.pdf](https://github.com/savushkin-r-d/EasyEPLANner/files/5327858/MA8694-8791-8696-IODD-file-EU-ML.pdf).

dimabeel, 2021-05-21:
Подтип - VC_IOLINK.

dimabeel, 2021-05-21:
Изделие - "BURKERT.8694"

PR #548 (2021-05-26): Added Burkert analog valve
Fixes #514.

Добавлен аналоговый клапан Burkert. Подтип - VC_IOLINK. Если стоит обычный клапан, подтип указывается VC

- [x] Добавить клапан Burkert
- [x] Указать параметры устройства
- [x] Указать экспортируемые параметры в базу каналов
- [x] Заполнить корректное изделие
- [x] Доработать юнит-тесты
- [x] Тестирование
- [x] Документация
closed 2021-05-26 by idzm