sav-oss-eplanner/EasyEPLANner #532: Доработка работы с базовыми параметрами

dimabeel, 2020-09-15:
Нужно сделать дополнительный механизм, который позволяет вводить параметры в поля не руками, а с помощью окна.
Предложено следующее:
1. Для булевых параметров перейти к выпадающему списку с "Да/Нет".
2. Для остальных параметров - реализовать окно по примеру настройки ограничений.

Соответственно, параметры принадлежат объекту, в котором они выставляются и окно должно содержать все параметры этого объекта и устройства проекта (если в это поле может быть добавлено устройство). Необходимо реализовать запрет на ввод более 1 параметра, если это не предусмотрено.

Скорее всего понадобится переработка механизма отображения объектов, чтобы GUI запрашивал уже готовое дерево и налаживал его на свою модель. Нужно будет переработать как часть редактора, так и окно "Операции".

idzm, 2020-09-15:
Тут спешить не надо - окно не всегда хороший вариант. Можно, вначале, и строковый вариант доработать - например, авто-дополнение строки с подсказкой для уже имеющимся вариантам после ввода первых двух символов и т.п. Переход в новое окно и потом возврат назад требует дополнительных затрат времени от пользователя (инженера по автоматизации) - это также надо учитывать.

dimabeel, 2020-09-15:
Кликать мышкой всегда проще, чем вводить на клавиатуре. Все равно остается вариант ввода вручную. Например, если надо добавить устройство, которое начинается PLACE1OBJECT1V1, и единственное отличие будет в номере клапана. Это мартышкин труд. Пользователю все равно придется смотреть на ФСА, чтобы запомнить, что он ищет. А в окне сразу весь список, можно будет если что свернуть (надо прорабатывать). С параметрами такая же песня, если пользователь знает только русскоязычное имя параметра? Автодополнение - не панацея. 
Плюс такую реализацию предложили пользователи.

idzm, 2020-09-15:
Я вижу автора - Умецкий Дмитрий, а это - разработчик, а не пользователь. Поэтому ещё опросим остальных пользователей для более полного мнения.

idzm, 2020-09-15:
@pupenasan, @KUZMA-1989, @AndreyNemenkov, @painterlv - что скажете по данному вопросу (описан ранее - в комментарии к запросу и в диалогах)?

pupenasan, 2020-09-15:
Не понял контекста. Если вобще о том, как проще вводить значение, например задания, то виртуальная клавиатура сразу даст возомжность вводить так, как юзеру удобнее. Если заданий много (рецепт) то может есть смысл делать только текстовый ввод. Опять же, для меня мало инфы, чтобы высказать свое мнение.

dimabeel, 2020-10-16:
Сейчас устройства или сигналы можно нажимать на схеме и они запишутся. Надо добавить чтобы они отображались в окне, и также отображались параметры объекта. В остальном все ОК.
Система должна определить, если можно добавлять параметр - показывать только параметры, если только устройства - устройства и т.д.
Для булевых (Да/Нет) параметров переделать нынешнюю реализацию, чтобы можно было выбирать не двойным кликом, а через комбобокс. Подумать, может как-то совместить эти две реализации вместе.

Для номеров операций оставляем как и есть т.к список операций там рядом и проще будет оставить так как есть.

Пользователю будет выводиться весь пул устройств и все параметры объекта в окне "Устройства" или "Операции". Ещё не решено.

PR #601 (2020-10-28): Update work with base parameters in object
Fixes #532

Цитата из issue:
> Сейчас устройства или сигналы можно нажимать на схеме и они запишутся. Надо добавить чтобы они отображались в окне, и также отображались параметры объекта. В остальном все ОК.
> Система должна определить, если можно добавлять параметр - показывать только параметры, если только устройства - устройства и т.д.
> Для булевых (Да/Нет) параметров переделать нынешнюю реализацию, чтобы можно было выбирать не двойным кликом, а через комбобокс. Подумать, может как-то совместить эти две реализации вместе.
> 
> Для номеров операций  [... cut]