sav-oss-eplanner/EasyEPLANner #545: Добавить поле для устройств, которые перед включением операции необходимо проверить на аварийное состояние

AlexandraNovik, 2020-09-23:
В текущем варианте проверяются устройства, которые прописаны в шагах, но логика обработки некоторых устройств может быть прописана в коде. Их состояние  на момент включения тоже нужно проверить. Также можно указать для проверки датчики, счетчики, мы ими не управляем, но опрашиваем. И если устройство в аварии, то операцию включать нельзя.

dimabeel, 2020-11-11:
@AlexandraNovik, то есть, указать в операции перечень устройств, которые будут всегда проверяться перед включением операции? Это могут быть любые опрашиваемые устройства в т.ч. устройства, логика которых прописана в коде?
Устройства, логика которых прописана в коде, должны проверяться независимо от того, будут ли эти устройства прописаны для проверки в Eplanner?

AlexandraNovik, 2020-11-11:
Да, возможно, датчик, который мы проверяем в коде неисправен, или мотор, которым управляем. Операцию включать нельзя. Насколько я понимаю, если устройство прописано в полях Включить, Выключить и т.д.  и есть ошибка этого устройства, то операция не включится.

dimabeel, 2020-11-11:
Да, они проверяются. 

А что по поводу последнего моего вопроса? Устройства, которые есть на ФСА, но они включаются и выключаются через код. Если они не прописаны нигде (даже если добавить поле "Проверяемые устройства"), такое же тоже предполагается, что надо проверить?

AlexandraNovik, 2020-11-11:
В этом поле "Проверяемые устройства" можно указать для проверки устройства, которые управляются через код.

idzm, 2020-12-11:
@AndreiSavushkin, @KUZMA-1989 - что скажете по внедрению данной функциональности?

AndreiSavushkin, 2020-12-11:
Нужно посмотреть стандарт. Описанное очень похоже на состояние "Включение"

idzm, 2020-12-14:
После обсуждения решили, что для инженера по автоматизации данная функциональность не нужна (ему хватает включаемых/выключаемых устройств). А инженер-программист АСУТП  так как пишет код, может данную функциональность реализовать самостоятельно.

idzm, 2021-07-21:
Будет реализовано, так как появится возможность контролировать устройства, подключаемые по IO-Link, на аварийные состояния.

PR #871 (2021-07-22): Add checked devices action
Fixes #545
closed 2021-07-22 by idzm