sav-oss-eplanner/EasyEPLANner #546: Использование временных параметров

AlexandraNovik, 2020-09-23:
В текущей версии есть возможность в шаге указать время его выполнения через параметр. По умолчанию данные параметры задаются в секундах. Но в некоторых случаях время шагов в объекте может измеряться в миллисекундах или минутах. Можно было бы использовать параметр с нужной размерностью, например, минуты, но указывать для операции дополнительный временной коэффициент для пересчета.  Коэффициент также может быть параметром. Для одного объекта используются одинаковые единицы времени.
Если используются секунды, то умолчанию коэффициент 1000. Для миллисекунд коэффициент 1, для минут 60000.

idzm, 2020-12-11:
@KUZMA-1989, @AndreiSavushkin - нужно ваше мнение по данному предложению.

AndreiSavushkin, 2020-12-11:
0.001 - 1 мс. Чем не задание миллисекунд?

AlexandraNovik, 2020-12-11:
Мы выводим параметры для последующего задания человеком: "Время шага N, сек". Если шаг в миллисекундах, ему нужно будет задать, например, 0.78 секунд, а если в часах, то 7200 секунд?

AndreiSavushkin, 2020-12-11:
Да, именно так.

idzm, 2020-12-14:
Так как время цикла управляющей программы в текущей реализации составляет ~200 мсек, не имеет смысла задавать временные параметры в миллисекундах - минимальное задание составляет 1 секунда (этого хватает для решения текущих задач). Если требуется выполнение каких-то блоков с другим быстродействием, необходимо реализовывать это на другом уровне (например, в виде дополнительной программы в PLCNEXT ENGINEER).

closed 2020-12-14 by idzm

AndreiSavushkin, 2020-12-14:
Могу с уверенностью сказать, Что шаг 3 секунды и 2.5 секунды на реальном объекте длится разное время