sav-oss-eplanner/EasyEPLANner #547: Обновление документации

AlexandraNovik, 2020-09-23:
Есть стандартные функции работы с колонной. Необходимо описание, как они работают, также описание в каких случаях какие цвета загораются. И что необходимо, если колонна отличается от стандартной (зеленый, желтый, красный, звук).

idzm, 2020-12-14:
Сигнальная колонна будет реализована в виде устройства (control module) со своей функциональностью. @AndreiSavushkin, @KUZMA-1989 - как назвать данное устройство?

AndreiSavushkin, 2020-12-14:
Должен задаваться тип красной лампы и звукового сигнала (выход либо постоянный либо мигающий)

dimabeel, 2021-05-18:
Можно написать устройство HLA - сигнальная колонна с сиреной. Всегда в ней будет лампа красная и сирена. Если подтип не указан, то будет только 2 DI сигнала (на красную лампу и сирену). Это устройство будет привязываться как оборудование к юниту.
Подтипы:
1. Не указан (красная лампа + сирена) - 2 DI
2. R_Y (красная лампа + желтая лампа + сирена) - 3 DI
3. R_G (красная лампа + зеленая лампа + сирена) - 3 DI
4. R_Y_G (красная лампа + желтая лампа + зеленая лампа + сирена) - 4 DI

Потом можно будет легко добавить IO-Link сигнальную колонну. 

Над свойством типа красной лампы и звукового сигнала надо ещё подумать

AndreiSavushkin, 2021-05-18:
DI? ГОСТ 29149-91 Цвета световой сигнализации и кнопок - в этом документе все описано. В некоторых физических устройствах мигание красной лампы заложено в непосредственно устройство (для этих устройств DO на красный канал должен быть постоянный), а для некоторых - импульсный (о чем я и просил 14.12.2020). Тоже самое касается и для звукового сигнала

idzm, 2021-05-18:
Сигнальная колонна. Служит для уведомления оператора о событиях.
Используется три лампочки – зелёная, жёлтая, красная и сирена:
     1. Зелёный цвет – есть активная операция (связанного аппарата или агрегата).
     2. Мигающий красный цвет – авария, тревога (не подавленная). Частота мигания 2 Гц. Дополнительно включается сирена, которую можно отключить проекта (клик по значку) либо она также отключится после подтверждения тревог, пропадания аварий.
     3. Мигающий жёлтый свет – неподтверждённое сообщение. Частота мигания 0.5 Гц.

idzm, 2021-06-18:
Поля для данного устройства:
- `M `- ручной режим;
- `ST `- состояние.

PR #825 (2021-06-24): Add light column device (HLA)
Fixes #547.

Добавить также это новое устройство в генератор заглушек для межпроектного обмена!
PR #861 (2021-07-01): Add light column device (HLA) properties
Fixes #547.
PR #863 (2021-07-02): Add HLA control module description
Fixes #547.
closed 2021-06-24 by idzm