sav-oss-eplanner/EasyEPLANner #551: Поле Мойка

AndreiSavushkin, 2020-09-24:
Необходимо переименовать узел Мойка в Устройства. 
Добавить функционал:
1 Реализовать комплекты сигналов (аналогично группам при флипе клапанов). 
2 В поле "Устройства" реализовать возможность включения любого аналогового устройства (помимо насосов с частотным управление, это также является AO, VC). Реверс - только моторы с частотным управлением.
Остальной функционал оставить прежним
[image]

idzm, 2020-11-03:
@AndreiSavushkin по количеству значение для полей: DI и частота насосов - может быть только одно значение?

AndreiSavushkin, 2020-11-04:
Нет. Частота - для каждого насоса может быть своя. К тому же, может быть и насос и клапан регулирующий.

PR #605 (2020-11-09): Modify "Wash" cell to "Devices" cell in Action
Fixes #551.
closed 2020-11-09 by idzm