sav-oss-eplanner/EasyEPLANner #561: Модернизация обмена сигналов

AlexandrBehunkov, 2020-10-13:
При обмене сигналов между проектами у  объекта "Приемник" необходимо добавить следуующее:
init_gateways(remote_gateways) в function user_init()
eval_gateways(remote_gateways) в function user_eval()

dimabeel, 2020-10-15:
@idzm , что если, отсутствует файл shared.lua, но будут вызваны эти функции, какой результат?

idzm, 2020-10-15:
Можно часть функций вынести в системные скрипты. @rartem - что скажешь?

dimabeel, 2020-10-20:
После беседы с @rartem были подведены определенные итоги по этому функционалу.

В файлы **main.plua** и **main.modbus_srv.lua** не предназначена автоматическая запись в виду того, что эти файлы предназначены и для пользовательского кода тоже.

Вынести в системные скрипты инициализацию **remote_gateways** можно, но это надо вычищать в зависимости от версии скриптов, файлы проектов. Потому, что если изменить все файлы проектов под новую версию скриптов, то обмен по клиентской части может не работать из-за старой версии скриптов в проектах и отсутствия нужных функций в **main.plua**. Единственное возможное хорошее решение - надо в скриптах писать обработчики, которые предотвращают повторный вызов функции в **init()** (_вызывается один раз_) и **eval()** (_вызывается циклически_).

Аналогичная проблема касается и **modbus_srv.lua**. При отсутствии **shared.lua**, автоматически сгенерированный код сервера будет ругаться, что ничего не нашел, поэтому нужно как-то обработать эту ситуацию в скриптах, чтобы автоматически включался сервер (_и предотвращался повторный вызов функции_) и в файл **modbus_srv.lua** записывался только пользовательский код.
@idzm , @AlexandrBehunkov - ваши предложения?

PR #588 (2020-10-20): Automatic run modbus server on controller
Fixes #561.

https://github.com/savushkin-r-d/ptusa-Lua-sys/pull/283 - #561.
Дополнительно здесь резолвится автоматический запуск модбас сервера на контроллере. В новых проектах автоматически генерится **shared.lua** (_с пустым описанием_) и **modbus_srv.lua**
closed 2020-10-20 by idzm