sav-oss-eplanner/EasyEPLANner #578: Межконтроллерное взаимодействие: ошибка в настройка PAC

KUZMA-1989, 2020-10-15:
В настройке PAC одном случае при открытии  у меня фиксированное значение IP адреса в источнике, второй раз, когда открываешь, то в приемнике
[image]
[image]

dimabeel, 2020-10-15:
@KUZMA-1989 , если попереключать между собой источник-приемник не меняется?

KUZMA-1989, 2020-10-16:
> @KUZMA-1989 , если попереключать между собой источник-приемник, адрес не меняется? Или пощелкать по именам других проектов. В источнике все IP-адреса должны быть одинаковы т.к это адреса которые будут записаны в клиенты (это адрес сервера), чтобы обращаться к серверу

Я открываю таблицу, и в зависимости, что ты первым нажмешь источник или приемник, то тот и оказывается с фиксированным адресом

dimabeel, 2020-10-16:
Все перепроверил, нужно больше информации. При каких условиях это происходило либо воспроизвести ошибку, чтобы я видел.

Проверил при создании с нуля обмена, при редактировании существующего, в различных режимах. IP-адреса источника всегда фиксированы, а приемника - разные. На картинке действительно косяк, но не могу его воспроизвести.

PR #584 (2020-10-16): Fix error with PAC settings edit mode changing
Fixes #578.

Не было обработки, когда изменяется с уже выбранного значения на это же значение (особенность событий).
closed 2020-10-16 by idzm