sav-oss-eplanner/EasyEPLANner #591: Имена параметров технологических объектов

KirillSimatic, 2020-10-21:
Составить список всех имен используемых для параметров технологических объектов
Добавить в документ "Руководство - разработка проекта автоматизации на SCADA EasyServer + CAD Eplan Electric P8"

dimabeel, 2020-10-21:
@idzm, где мне взять список всех параметров?

KirillSimatic, 2020-10-21:
Исходя из  запущенных  проектов . Нужно разработать кодировку формирования имен для параметров чтобы потом выбирать их  из  списка (как говорил  Дмитрий Иванюк).

dimabeel, 2020-10-21:
Список параметров в проектах может быть излишний на первых парах. Уже существующие параметры можно либо собрать по всем проектам (как ты говоришь), либо взять их в скриптах, потому что работа с этими параметрами может быть прописана в main.plua (написана пользователем), но автоматически обрабатываются только те параметры, которые прописаны в скриптах. Теряется смысл автоматической генерации, если эти параметры не будут обработаны в скриптах. Ещё проработается этот вопрос, как лучше сделать

KirillSimatic, 2020-10-21:
Инженер по автоматизации должен работать с скриптами ?

dimabeel, 2020-10-21:
Конкретно с теми скриптами, которые грузятся в контроллер, работает в основном @idzm, но если есть предложения - может любой, но все правки проходят через @idzm. Остальные файлы - **main.plua** и другие (_которые поддерживают пользовательский код_) могут править и инженера, но вроде как сейчас этим занимается @AlexandrBehunkov. Я код для проектов не пишу.

closed 2020-10-21 by KirillSimatic