sav-oss-eplanner/EasyEPLANner #596: Модуль оборудования

KirillSimatic, 2020-10-21:
Реализовать возможность повторного использования модуля оборудования. Для наших  проектов возможно создать :  модуль нагрева, модуль охлаждения (регулирующий клапан , бачок с ледяной водой), регулятор потока , мешалка .
[Eplan MO.pdf](https://github.com/savushkin-r-d/EasyEPLANner/files/5415649/Eplan.MO.pdf)

dimabeel, 2020-11-11:
Мы в принципе к такому и идем, только в разрезе базового объекта. Нынешняя генерация будет дополняться автоматической генерацией операций, параметров, шагов. Это в работе. Будут ссылки на этот issue в процессе разработки.

Если есть уже готовые объекты - можно использовать механизм экспорта и импорта объектов в надстройке. Это позволит избежать множества ручной работы. Вот документация по экспорту и импорту: [ТЫК](https://github.com/savushkin-r-d/EasyEPLANner/tree/master/docs/user_manual#74-%D0%AD%D0%BA%D1%81%D0%BF%D0%BE%D1%80%D1%82-%D0%BE%D0%B1%D1%8A%D0%B5%D0%BA%D1%82%D0%BE%D0%B2-%D0%B8%D0%B7-%D1%80%D0%B5%D0%B4%D0%B0%D0%BA%D1%82%D0%BE%D1%80%D0%B0-%D0%B8-%D0%BE%D0%B1%D1%80%D0%B0%D1%82%D0%BD%D1%8B%D0%B9-%D0%B8%D0%BC%D0%BF%D0%BE%D1%80%D1%82)

dimabeel, 2021-01-05:
Пути:
1. Написать свою кнопку, на эту кнопку повесить интеракцию, и вставлять макросы пока активна кнопка.
2. Повесить интеракцию на кнопку "Вставить макрос", и по этой кнопке вызывать интеракцию. Завершать интеракцию после вставки макроса (_наверное какое-то из событий OnSmth_).

dimabeel, 2021-01-06:
Захват кнопки:
1. Поиск окна Eplan через PInvoke.
2. Установка хука на окно
3. Искать WMCOMMAND кнопки
4. По нажатию на кнопку запускать интеракцию (будет запущено две, или какая-то перекроет? Что будет, если перекроет?)
5.  TODO

После интеракции вызывалось событие UpdateConnections - надо его тоже как-то видимо вызвать или что-то в этом духе. Потому что не понятно пока, как будут обрабатываться интеракции и могут ли быть запущены одновременно две, например.

dimabeel, 2021-01-12:
Ответ от русской службы поддержки был получен. Можно переопределить Interaction.
https://www.eplan.help/en-US/infoportal/content/api/2.9/Interactions.html#i-heading-overriding-default-interactions-

dimabeel, 2021-01-13:
При вставке макроса мы получаем вставленные объекты в виде массива, причем это и линии, и функции объектов. Нужно либо придумать свою функцию, которая будет являться описательной (_на её основе будет что-то генерироваться и др._), либо писать в каком-то из объектов макроса.

idzm, 2021-03-10:
Базовый функционал реализован в #712. Дальнейшая доработка будет после реализации доступа к объектам, параметрам
 и другим элементам по именам. Сейчас в основном доступ по индексам.

closed 2021-03-10 by idzm