sav-oss-eplanner/EasyEPLANner #604: EPLAN Improvment

PLIeplan, 2020-10-28:
Добрый день.
Прошу подготовить демо-проект в zw1 для тестирования функционала модуля ePlaner и оценки возможностей интеграции его в стандартный функционал платформы.
Could you prepare demo-project ZW1 for API Module ePlaner testing and estimating integration it into the plaform.
Polunin Igor
Eplan Team

dimabeel, 2020-10-30:
https://github.com/savushkin-r-d/EasyEPLANner/releases/tag/2020.10
Ссылка на релиз. Там в **.zip** архиве и файл **.zw1**, и файлы проекта (_lua_), чтобы их можно было подгрузить при загрузке проекта и посмотреть как все сделано в редакторе технологических объектов.

idzm, 2020-11-04:
@PLIeplan - по текущему демо-проекту есть вопросы?

PLIeplan, 2020-11-06:
Добрый день. Пока нет. Сейчас в процессе тестирования

PLIeplan, 2020-11-12:
Добрый день. Коллега написал, что не смог найти EPLAN.EplAddin.EasyEPlanner.dll, указанный в readme. Программист тоже не смог скомпилировать в VStudio, сославшись на то, что не хватает ссылок.
Проект ZW1 рабочий.
Можете сами скомпилировать и ссылку прислать на EPLAN.EplAddin.EasyEPlanner.dll?

dimabeel, 2020-11-12:
Обновил релиз. Пройдите по той же ссылке ([дублирую](https://github.com/savushkin-r-d/EasyEPLANner/releases/tag/2020.10)).

Выберите архив "**EasyEPlannerBinaries.zip**" (_см. картинку_).
[image]

Архив содержит в себе полную рабочую копию надстройки в том виде, в котором мы её используем, без учета файла конфигурации, который генерируется автоматически, на **12.11.2020**. В том числе, в архиве есть нужная dll-ка.

Архив внутри:
1. папка **CMD** - команды для запуска тестирование и проверки наличия LUA в системе (_по-умолчанию отключено_). 
2. папка **Lua** - содержит рабочие скрипты, которые мы используем для загрузки данных из Lua-файлов проектов, а также скрипты, которые служат хранилищами данных.
3. папка **LuaInterface** - содержит библиотеки для коммуникации между C# кодом и Lua.
4. **Aga.Controls.dll** - открытая UI библиотека. Используем в окнах "Устройства" и "Операции".
5. **EPLAN.EplAddin.EasyEPlanner** - сама надстройка.
6. **KopiLua**.dll - библиотека из папки LuaInterface.
7. **LuaInterface.dll** - библиотека из папки LuaInterface.
8. **ObjectListView** - открытая UI библиотека. Используем в окне "Редактор технологических объектов".
9. библиотеки **Spire.*.dll** - открытые библиотеки для работы с документами. Используем для генерации xlsx отчета по проекту.

PLIeplan, 2020-11-19:
Спасибо. Модуль заработал, проект lua прогрузился. Сейчас будем разбираться с функционалом конфигурирования структуры.

PLIeplan, 2020-12-17:
Дмитрий, спасибо. Пока все получается. Читаем инструкцию и параллельно тестируем.

idzm, 2021-03-30:
Тестовый демонстрационный проект был подготовлен (актуальная версия находится вместе с текущим релизом). Если появятся замечания/предложения/отзыв, то это можно оформить в виде нового вопроса (issue).

closed 2021-03-30 by idzm