sav-oss-eplanner/EasyEPLANner #628: Доработать базовые шаги

dimabeel, 2020-11-18:
Нужна возможность задавать базовые шаги в состояниях "Пауза" и "Стоп" в операции.

dimabeel, 2020-11-20:
Надо продумать этот вопрос, поскольку в нынешней реализации состояние не учитывается и предполагается, что мы всегда работаем с состоянием "**Выполнение**". Это реализовано в том числе и в скриптах, поэтому нельзя в лоб добавить поддержку состояний в базовое описание, это приведет к проблемам обработки в скриптах.

idzm, 2021-03-23:
Предлагается добавить дополнительные поля, для указания состояния. Вот образец: 

```Lua
prg.tank_a1.steps =         --Шаги операций.
    {
    WASHING_CIP =
        {
        RUN =
            {
            MEDIUM_CHANGE = 5,
            DRAINAGE = 7,
            },
        PAUSE =
            {
            WAITING_SIGNALS = 1,
            },
        STOP =
            {
            SWITCHING_OFF_SIGNALS = 1,
            SWITCHING_OFF_DEVICES = 2,
            }
        }
    }
```

@AlexandraNovik - пойдёт такой вариант?

AlexandraNovik, 2021-03-23:
Да, я таблицей такого вида и пользуюсь. Вопрос, если в каком-то состоянии шаги не предусмотрены?

dimabeel, 2021-03-23:
Записи PAUSE, RUN или STOP в данном случае не будет

closed 2021-03-30 by idzm

AndreiSavushkin, 2021-04-13:
Реализация в таком виде привело к тому, что во множестве реализованных проектов в местах, где прописана логика работы в main, необходимо вручную изменять код!!! Реализуйте так, чтобы новый функционал не приводил к необходимости ручных правок в большом множестве действующих проектов
> ```lua
>    RUN =
>             {
>             MEDIUM_CHANGE = 5,
>             DRAINAGE = 7,
> ```

PR #796 (2021-03-30): Add base steps into PAUSE and STOP state
Fixes #628.

Tasks:
- [x] Add LuaName field in State class.
- [x] Block use base step in "Во время операции" step
- [x] Load base step depending on state
- [x] Save base step depending on state in prg.lua
- [x] Correct export of channel base
- [x] ~Possibility to add new states without adding much code. States have to be sorted and be independent from their index.~
- [x] Support all old functionality and use it in another states
- [x] Refactoring
- [x] Unit-tests
- [x] ~Documentation~