sav-oss-eplanner/EasyEPLANner #674: Возможность изменения типа базового объекта

dimabeel, 2020-12-16:
Цитата:

> Функционал смены типа объекта необходимо реализовывать - тип объекта может изменяться при модернизации производства. Нужно делать в новом каком-то окне с отображением изменений, которые произойдут. Нужно максимально оставлять имеющийся функционал. Зачем же например удалять какие-либо операции? - удаляйте только тип этой базовой операции, а сама операция, шаги должна остаться.

Подумать над тем, как это правильно сделать. Были предложения, чтобы при смене базового объекта показывалось, какие объекты удаляются, какие добавляются и какие не изменяются. Объекты, которые не изменяются - не трогать (_меняется базовый владелец, типа такого_)

dimabeel, 2021-02-12:
Этот запрос будет закрыт в несколько этапов. Первый этап - сбрасывать базовый объект в состояние "неопознанный объект".

idzm, 2021-12-30:
На текущий момент это реализовано через сброс базового объекта (в состояние "неопознанный объект") и выбор нового.

PR #758 (2021-02-16): Reset base tech object through editor
Partial closes #674.

- [x] Функционал
- [x] Рефакторинг
- [x] ~Юнит-тесты (если будет не в легаси классе)~
- [x] Ручное тестирование
- [x] Документация
closed 2021-12-30 by idzm