sav-oss-eplanner/EasyEPLANner #702: Добавление операции "Отделение сыворотки (по оборотам)" для базового аппарата

idzm, 2021-01-05:
Lua-имя - `WHEY_SEPARATION`.

Параметры:
1. Максимальное время операции - `MAX_TIME`.

dimabeel, 2021-01-05:
@idzm, у нас уже есть отделение сыворотки (нагрев) - _WHEY_SEPARATION_PRE_. Надо дать наверное другое название, чтобы учитывалось что по оборотам.

idzm, 2021-02-21:
Также необходимо добавить следующие параметры:
    1. **CURRENT_STEP** - текущий шаг, номер параметра;
    2. **STEP1_SPEED** - скорость перемешивания для шага 1 (обороты/мин), номер параметра;
    3. **STEP1_DIRECTION** - направление перемешивания для шага 1, номер параметра;
    4. **STEP1_COUNT** - количество оборотов  для шага 1, номер параметра;
    5. **STEP1_WAIT** - время ожидания после перемешивания (мин), номер параметра;
    6. **HEAT_TEMPERATURE** - температура подогрева, номер параметра.

dimabeel, 2021-02-22:
@idzm, может быть не номер параметра надо указывать, а его имя?

idzm, 2021-02-22:
Все правильно, поправить описание нужно.

PR #772 (2021-02-22): Add new params to whey separation turnover
Fixes #702
closed 2021-02-22 by idzm