sav-oss-eplanner/EasyEPLANner #718: Возможность привязки сигналов в поля доп.свойств

KUZMA-1989, 2021-01-19:
Сделать, чтоб можно было в допсвойствах устройств аппаратов, агрегатов можно было привязывать сигналы: для дискретных устройств- привязка DI, для аналоговых устройств - AI

KUZMA-1989, 2021-01-19:
Это расширит функционал, если мы работаем с обменом сигналами с партнерами

dimabeel, 2021-01-25:
Для этой цели лучше использовать виртуальные устройства (нужно добавить), и передавать устройства через межпроектный обмен сигналами. На данный момент есть виртуальные устройства: FQT, LT, LS. 
Нужен список виртуальных устройств и я его добавлю. Такое взаимодействие в мойках каких-то реализовано.

@rartem будет работать же?

rartem, 2021-01-25:
Должно работать

dimabeel, 2021-01-25:
@KUZMA-1989, мне нужен список устройств, которые мы сделаем виртуальными, помимо FQT, LT, LS.

KUZMA-1989, 2021-01-25:
Устройства могут быть виртуальными все,которые мы используем

PR #868 (2021-07-16): Add virtual subtypes for all remaining devices
Fixes #718

For all devices except Y, DEV_VTUG and C (PID's)

- [x] Unit-tests for functionality
- [x] Functionality
- [x] Refactoring
- [x] Manual testing
- [x] Documentation
- [x] Wait for adding this functionality in PLC program
closed 2021-07-16 by idzm