sav-oss-eplanner/EasyEPLANner #766: Механизм описания параметров базовых объектов и автоматической настройки

dimabeel, 2021-02-16:
При вставке объекта (_нового_) автоматически нужно заполнять параметры базового объекта, если они есть. Учесть, что у объектов могут быть параметры, которые общие (_есть во всех или почти во всех объектах)_. Параметры описываем там, где описание базовых объектов.

dimabeel, 2021-02-17:
```LUA
base_tech_objects = function()
return
    {
        automat = {
            name = "Автомат",
            s88Level = 2,
            baseOperations = { },
            basicName = "automat",
            equipment = { },
            aggregateParameters = { },
            bindingName = "automat",
            parameters =
            { 
                global_parameters.luaName,
                global_parameters.neLuaName,
            },
        }
    }
end

global_parameters =
{
    luaName = { name = "parName", luaName = "luaName" },
    neLuaName = { name = "nepar", luaName = "neLuaName" },
}
```

Предлагаю следующий формат описания параметров. Так как мы используем только **float** параметры и в редких случаях **runtime float** (_обычно МСА_), то не вижу смысла делать **runtime float**, поскольку они часто копипастятся и все. 

Такой формат позволит убрать дублирование в объектах (_можно потом распространить на все остальные аспекты, такие как параметры пид и прочее_), но появится дублирование Lua имени (_см код_), потому что при обращении к переменной **global_parameters.luaName** мы получим сразу таблицу, и никак не сможем получить имя переменной. В связи с этим, дублируется **luaName** в таблице описания параметра.

@idzm, как тебе вариант?

idzm, 2021-02-17:
Может быть. Имя поля таблицы тоже можем получить, только зачем у нас тут два поля?

idzm, 2021-02-17:
neLuaName - зачем это поле?

dimabeel, 2021-02-17:
@idzm чтобы обратиться по переменной **neLuaName** в описании параметров объекта. Что-то вроде **global_parameters.neLuaName**. В итоге я получу таблицу, и не смогу получить **neLuaName**. Поэтому дублируется.

dimabeel, 2021-02-17:
```LUA
table = {
    global_parameters.neLuaName,
    global_parameters.luaName,
}
```

тоже самое, что:
```LUA
table = {
    [ 1 ] =
    {
         luaName = "neLuaName",
        -- Опущена остальная часть
    },
   [ 2 ] =
   {
       luaName = "luaName",
       --Опущено остальное
   },
}
```

PR #767 (2021-02-19): Automatic insert parameters when inserting new object
Closes #766.

- [x] Функционал
- [x] Рефакторинг
- [x] Юнит-тесты (если не легаси код, или не нужно перерабатывать очень много)
- [x] Ручное тестирование
- [x] Документация (в файле описания т.к пользователю это не надо)
closed 2021-02-19 by idzm