sav-oss-eplanner/EasyEPLANner #776: Описание базовой операции "Откачка сыворотки"

idzm, 2021-02-23:
Необходимо добавить для аппарата описание базовой операции "Откачка сыворотки". Lua-имя - `WHEY_PUMPDOWN`. Свойства:
1. `UPPER_BUTTON`, `UPPER_STEP` - имя параметра и шаг для откачки из верхней части танка;
2. `MIDDLE_BUTTON`, `MIDDLE_STEP` - имя параметра и шаг для откачки из средней части танка;
3. `LOWER_BUTTON`, `LOWER_STEP` - имя параметра и шаг для откачки из нижней части танка.

dimabeel, 2021-02-23:
@idzm, базовые шаги тоже надо указать. Зачем тебе параметры UPPER_STEP, MIDDLE_STEP, LOWER_STEP если мы можем просто указать базовые шаги?

dimabeel, 2021-02-23:
Фраза "танк" тут лишняя в описании параметров и шагов. Мы редактируем объект танк, поэтому мы уже знаем, что это откачка из танка. Может написать просто "Откачка из верхней части"?

dimabeel, 2021-02-23:
[image]

idzm, 2021-02-23:
> Фраза "танк" тут лишняя в описании параметров и шагов. Мы редактируем объект танк, поэтому мы уже знаем, что это откачка из танка. Может написать просто "Откачка из верхней части"?

Да, пойдёт.

idzm, 2021-02-23:
> @idzm, базовые шаги тоже надо указать. Зачем тебе параметры UPPER_STEP, MIDDLE_STEP, LOWER_STEP если мы можем просто указать базовые шаги? Или я не так понял описание?

Параметр нужен - по его изменению мы включаем/отключаем соответстующий шаг.

dimabeel, 2021-02-23:
Тогда и шаги нужны. Указав номер шага для операции, у которой базовая не указана - не будет сгенерировано описание

dimabeel, 2021-02-23:
Принято решение убрать параметры и оставить только шаги.

PR #777 (2021-02-23): Add WHEY_PUMDOWN operation with steps and properties
Fixes #776.
closed 2021-02-23 by idzm