sav-oss-eplanner/EasyEPLANner #780: Добавление полей в списках устройств шагов, возможности перехода по шагам

AlexandraNovik, 2021-02-23:
При реализации некоторых нетиповых объектов (картонажники, некоторые делители) возникают сложности в реализации.
Эти объекты являются сложными механически, и их физическое исполнение (количество и последовательность устройств) зависит от места реализации, продукта, задачи, которую необходимо выполнить. Но с логической точки зрения эти объекты представляют собой последовательность переключаемых шагов, в которых для переключения в следующих шаг нужно проверить дополнительные условия и устройства. При этом количество шагов в объектах всегда различно. 
Хотелось бы иметь возможность описать логику таких объектов в еплане, а не создавать в каждом проекте несколько уникальных объектов. Так можно было бы сгруппировать задаваемые в операциях/шагах устройства и добавить некоторые поля. Например:

1. Работа устройств:
- Включать
- Включать реверс
- Выключать
- **Контроль включения**
- **Контроль выключения**
- **Контроль включения параметров (bool)**
- **Контроль выключения параметров (bool)**
2. Дополнительные устройства
- Сигналы для включения
- Мойка
- Верхние седла
- Нижние седла
- Группы DI-DO
- Группы AI-AO
3. Переключение шагов
- **Время шага**
- **Шаг по условию**
- **Шаг по времени**
- **Переход в аварию**

**Контроль включения**
Задается список устройств, включение которых нужно проконтролировать. Это или сигналы, которые должны прийти к нам или включаемые нами устройства, приход ОС о включении которых мы должны проконтролировать.

**Контроль выключения**	
Задается список устройств, выключение которых нужно проконтролировать. Это или сигналы, которые должны сняться или выключаемые нами устройства, приход ОС о выключении которых мы должны проконтролировать.

**Контроль включения параметров**
Задается список параметров, включение которых нужно проконтролировать. Это параметры, которые нужно проконтролировать, что они установлены.

**Контроль выключения параметров**	
Задается список параметров, выключение которых нужно проконтролировать.
Это параметры, которые нужно проконтролировать, что они сброшены.
	
**Время шага**	
Время, по истечении которого происходит переход  в следующий шаг.

**Шаг по условию**	
Номер шага, в который переходим при корректной отработке устройств (все что нужно включилось/выключилось, нужные сигналы пришли/снялись).

**Шаг по времени**	
Номер шага, в который переходим по истечении заданного времени.

**Флаг «Выдача аварии»**
Установка флага, означающего, что необходимо выдать аварию и перейти в паузу если время шага вышло, а устройства не отработали должным образом.

**Логика обработки параметров, заданных в группе «Переключение шага»:**

Если задан только «Шаг по условию», то контролируем вкл./выкл. устройств («Контроль включения», «Контроль выключения») и вкл/выкл булевских параметров («Контроль включения параметров», «Контроль выключения параметров») и переключаем в него, как только они будут выполнены.  (Предполагается, что контролировать время в данной конфигурации не нужно).

Если задан только «Шаг по времени», то переключаем в него, когда вышло «Время шага». 
(Предполагается, что контролировать устройства в данном случае не нужно).

Если заданы и «Шаг по условию» и «Шаг по времени», то ждем пока выйдет «Время шага» и только тогда проверяем условия: 
Если условия выполнены, то переходим в «Шаг по условию»;
Если условия не выполнены, то: 
Если НЕ установлен флаг «Выдача аварии», то переходим в «Шаг по времени»;
Если установлен флаг «Выдача аварии», то выдаем аварию и переводим объект в «Паузу».

dimabeel, 2021-02-23:
@AlexandraNovik, а где это будет обрабатываться? В контроллере такой функциональности, как мне известно - нету

dimabeel, 2021-02-23:
Не могу понять, что значит "Включение" и "Выключение" параметров. Это как?

dimabeel, 2021-02-23:
"Время шага" и "Шаг по времени" - это уже есть. См скриншот
[image]

AlexandraNovik, 2021-02-23:
Знаю, что на данный момент есть возможность перехода по шагам по времени. Предлагаем это расширить, так как не хватает. И да, я знаю, что функциональности нет, но запрос же надо где-то первый оставить. Поля нужны в Еплане и их обработка в контроллере.

dimabeel, 2021-02-23:
Окей. Пусть тогда висит, я с Димой обсужу.
Остаются другие вопросы, по "Включение" и "Выключение "параметра

AlexandraNovik, 2021-02-23:
Да, обсудить нужно. Просто у нас много объектов, которые из-за контроля включенности и выключенности нужно описывать в коде (например, последовательное переключение цилиндров). Логика одинаковая просто устройства переключаем, а количество устройств разное, количество шагов разное.
Про параметры. Тоже проверка установлен он или нет (булевский работает по принципу флага). Иногда мы можем проверить датчик или сигнал (физическое устройство), иногда какое-то условие проверяется установленным флагом.

dimabeel, 2021-02-23:
Про параметры довольно специфически. Судя по описанию, он будет включаться всегда через код. Это нужна обработка скриптом (по аналогии с тем, что в АСУТП сейчас). Это что-то похожее на рецепт, если я правильно понимаю. Но с таким вот использованием придется постоянно что-то новое реализовывать. Потому что параметр Х подойдет только для одного случая, а для следующего уже придется писать алгоритм для параметра Y, и повезет, если Х и Y будут реюзабельными. Если я правильно понимаю.

idzm, 2021-02-23:
@AlexandraNovik - очень сложный вопрос. Необходимо разбить на запросы поменьше :-).

AlexandraNovik, 2021-02-23:
Хорошо, описали комплексную картину, как в итоге может получится. Можем описать частями, для поэтапной реализации отдельных функций.

dimabeel, 2021-09-14:
Основное реализовано, остальное реализуется в Lua коде.

closed 2021-09-14 by dimabeel