sav-oss-eplanner/EasyEPLANner #810: Добавление устройства "ПИД-регулятор"

idzm, 2021-04-16:
Необходимо добавить возможность описания устройства (control module) "ПИД-регулятор".
Тип устройства - **REGULATOR**. 

Вот  список параметров:

```C++
            k = 1,	///< Параметр k.
            Ti, 	///< Параметр Ti.
            Td,    	///< Параметр Td.
            dt,   	///< Интервал расчёта, сек.

            max, 	 ///< Мax значение входной величины.
            min, 	 ///< Мin значение входной величины.

            acceleration_time,	///< Время выхода на режим регулирования, сек.
            is_manual_mode,	///< Ручной режим, 0 - авто, 1 - ручной.
            U_manual,		///< Заданное ручное значение выходного сигнала.

            is_reverse,		///Обратного (реверсивного) действия", defaultValue = "false".
            is_zero_start,	///Нулевое стартовое значение", defaultValue = "true".

            k2,                  ///< Параметр k2.
            Ti2,                 ///< Параметр Ti2.
            Td2,                 ///< Параметр Td2. 

            out_max,             ///< Мax значение выходной величины.
            out_min             ///< Мin значение выходной величины.
```
Оборудование: - датчик и исполнительное устройство (control module):

```Lua
IN_VALUE = 'TE1', OUT_VALUE = 'VC1'
```

Задание задаётся в действии **"Устройства"** в **"Поле производительность"**. Это может быть параметр, либо **АI** или другой **ПИД** (control module).

idzm, 2021-04-16:
@KUZMA-1989, @AndreiSavushkin - что скажете?

dimabeel, 2021-04-19:
```LUA
is_reverse,        ///Обратного (реверсивного) действия", defaultValue = "false".
is_zero_start,	///Нулевое стартовое значение", defaultValue = "true".
```

Значение 0 - false
Значение 1 - true

AndreiSavushkin, 2021-04-20:
А к чему можно привязать выход регулятора?

dimabeel, 2021-04-20:
> А к чему можно привязать выход регулятора?

Выход регулятора привязывается к исполнительным устройствам или ко входу другого регулятора, для образования каскадного регулирования.

AndreiSavushkin, 2021-04-20:
> 
> 
> > А к чему можно привязать выход регулятора?
> 
> Выход регулятора привязывается к исполнительным устройствам или ко входу другого регулятора, для образования каскадного регулирования.

AO?

idzm, 2021-04-20:
Можно привязать к исполнительным устройствам (**AO**, **VC**, **M**) или ко входу другого регулятора (**R**).

PR #811 (2021-04-21): Add PID-regulator as Control Module
Fixes #810 

- [x] Добавление ПИД-регулятора как Control Module
- [x] Проверять записанные значения на корректность
- [x] Генерация базы каналов с новым ПИД-ом
- [x] Протоколирование Z и V
- [x] Рефакторинг
- [x] Юнит тесты
- [x] Документация
closed 2021-04-21 by idzm