sav-oss-eplanner/EasyEPLANner #906: Ошибки в полях при переносе объекта из пользовательского

KUZMA-1989, 2021-09-08:
Переводил проект BR1-Приемка из версии 2.2 на версию 2.9. Отобразились объекты из старого проекта в разделе "Пользовательский объект". Перенес их в группа агрегатов и аппаратов.
- [ ]  Автоматически заполнились поля "Оборудование" в танках, но только во всех танках были привязаны устройства из первого в списке танка (у меня был это TANK16).
- [ ] Для линий в объектах, тоже заполнились поля оборудование из первой в списке линии. 
- [ ] Во всех объектах в параметрах, которые были привязаны к операциям, поменялись значения в поле "операция" на -1.

dimabeel, 2021-09-08:
@KUZMA-1989 
1. В "Пользовательский объект" переход невозможен. Только в "Неопознанный объект".
2. Автоматическое заполнение оборудование предусмотрено всегда. Можно добавить ограничение, чтобы при изменении типа базового объекта этого не происходило, но тогда всегда придется записывать оборудование вручную (при изменении базового объекта)
3. По линиям - аналогично
3. Это должно было работать т.к не зависит от базовой функциональности

Какой проект? Пункт 3 я проверю, пункт 1 и 2 - так и задумано.

KUZMA-1989, 2021-09-08:
> @KUZMA-1989
> 
> 1. В "Пользовательский объект" переход невозможен. Только в "Неопознанный объект". При условии, что перевелся проект из 2.2 в 2.9
> 2. Автоматическое заполнение оборудование предусмотрено всегда, при вставке объекта (будь то замена типа, или вставка нового объекта такого типа). Можно добавить ограничение, чтобы при изменении типа базового объекта этого не происходило, но тогда всегда придется записывать оборудование вручную (при изменении базового объекта)
> 3. По линиям - аналогично
> 4. Это должно было работать т.к не зависит от базовой функциональности
> 
> Какой проект? Пункт 3 я проверю, пункт 1 и 2 - так и задумано.

Так в поле оборудование заполнилось оборудование не из того объекта, как это так задумано?

dimabeel, 2021-09-08:
@KUZMA-1989, не так прочитал по поводу оборудования

dimabeel, 2021-09-08:
Подтверждено. Разбираюсь

PR #907 (2021-09-08): Hotfixes from issue #906
Fixes #906
closed 2021-09-08 by skiftol