sav-oss-eplanner/EasyEPLANner #1075: Отображение в списке устройств моторов с типом M_ATV

KUZMA-1989, 2022-12-12:
Сделать отображение в списке устройств моторов с типом M_ATV, когда нажимаешь кнопку "Без привязки", чтоб можно было посмотреть какие моторы существуют в проекте. Во вложенном списке мотора , должен отобразиться IP адрес мотора

KirillGutyrchik, 2022-12-12:
Данный функционал уже реализован (#1027). Для отображения всех устройтсв, их параметров и свойств нужно нажать на кнопку "Отображать параметры и свойства" в окне дерева устройств .

https://github.com/savushkin-r-d/EasyEPLANner/blob/master/docs/user_manual/ReadMe.md#63-%D0%B4%D0%BE%D0%BF%D0%BE%D0%BB%D0%BD%D0%B8%D1%82%D0%B5%D0%BB%D1%8C%D0%BD%D1%8B%D0%B5-%D0%BF%D0%B0%D1%80%D0%B0%D0%BC%D0%B5%D1%82%D1%80%D1%8B

KUZMA-1989, 2022-12-12:
> Данный функционал уже реализован (#1027). Для отображения всех устройтсв, их параметров и свойств нужно нажать на кнопку "Отображать параметры и свойства" в окне дерева устройств . [...]

Отлично

closed 2022-12-12 by KUZMA-1989