sav-oss-eplanner/EasyEPLANner #1080: Орфографические ошибки в EasyEPLANner/docs/developer_manual/ReadMe.md

Pandem1r, 2022-12-13:
1. После модулей не должно быть "-", этот знак должен стоять между "ввода-вывода".
2.1.3 Каталог Lua
sys_io_modules.lua - скрипт с описанием **модулей-ввода вывода**, которые поддерживаются надстройкой.
Ссылка на пункт:  https://github.com/savushkin-r-d/EasyEPLANner/tree/master/docs/developer_manual#24-%D1%84%D0%B0%D0%B9%D0%BB%D1%8B-%D0%BF%D1%80%D0%BE%D0%B5%D0%BA%D1%82%D0%B0-eplanidletimemodule
 
 2. Должно быть "в течение 1 минуты", а не "в течении".
 2.2.2 Файл конфигурации модуля простоя
 maxChecksCount - количество успешных проверок подряд (когда система простаивает), после которого запускается механизм оповещения пользователя и затем отключения EPLAN, если пользователь не отреагировал **в течении 1 минуты**. Меняется руками.
 Ссылка на пункт:  https://github.com/savushkin-r-d/EasyEPLANner/tree/master/docs/developer_manual#24-%D1%84%D0%B0%D0%B9%D0%BB%D1%8B-%D0%BF%D1%80%D0%BE%D0%B5%D0%BA%D1%82%D0%B0-eplanidletimemodule
 
 3. Должно быть "какие-либо", а не "какие либо".
 2.4 Файлы проекта EplanIdleTimeModule
Проект достаточно прост. Закрывать EPLAN, если система простаивает какое-то время. Если открыты **какие либо** всплывающие окна в EPLAN, то проект не может быть закрыт и тогда процесс EPLAN уничтожается через Kill метод процесса.
Ссылка на пункт:  https://github.com/savushkin-r-d/EasyEPLANner/tree/master/docs/developer_manual#24-%D1%84%D0%B0%D0%B9%D0%BB%D1%8B-%D0%BF%D1%80%D0%BE%D0%B5%D0%BA%D1%82%D0%B0-eplanidletimemodule
   
    Ссылка на документ с ошибками:
    https://github.com/savushkin-r-d/EasyEPLANner/tree/master/docs/developer_manual#24-%D1%84%D0%B0%D0%B9%D0%BB%D1%8B-%D0%BF%D1%80%D0%BE%D0%B5%D0%BA%D1%82%D0%B0-eplanidletimemodule

KirillGutyrchik, 2022-12-14:
Исправьте кодировку ссылок как в 1-м пункте.
Сойздайте fork проекта, внесите изменения в файл и создайте pull-request.

Pandem1r, 2022-12-14:
> Исправьте кодировку ссылок как в 1-м пункте. Создайте fork проекта, внесите изменения в файл и создайте pull-request.

Сделал

PR #1082 (2022-12-16): [doc] Исправление орфографических ошибок
Fixes #1080.
Исправлены ошибки в документации.
closed 2022-12-16 by idzm