sav-oss-eplanner/EasyEPLANner #1090: Ошибки и опечатки в документации

1Nick10, 2022-12-20:
Слово "вырезать" некорректно взято в кавычки.
[image]
Отсутствует запятая перед "... в который...".
[image]
Опечатка в слове "их".
[image]

1Nick10, 2022-12-20:
Ссылка на документ: 
https://github.com/savushkin-r-d/EasyEPLANner/blob/master/docs/user_manual/ReadMe.md

KirillGutyrchik, 2022-12-21:
Необходимо создать fork проекта, внести изменения в файл и создать pull-request.

PR #1091 (2022-12-26): [doc] Исправление ошибок в документации
Fixes #1090 
Исправлено 3 опечатки.
closed 2022-12-26 by idzm