sav-oss-eplanner/EasyEPLANner #1126: Возможность отключать в настройках автозакрытие EPLAN при неактивности пользователя

idzm, 2023-03-01:
Реализовать возможность через настройки управлять работой модуля контроля активности пользователя.

idzm, 2023-03-01:
Возможно потребуется доработать функциональность - закрывать открытый проект, но не закрывать **EPLAN**. @AndreyNemenkov - что скажешь?

AndreyNemenkov, 2023-03-02:
> Возможно потребуется доработать функциональность - закрывать открытый проект, но не закрывать **EPLAN**. @AndreyNemenkov - что скажешь?
Я уверен, что лучше не закрывать проект вообще, мне не очень понравилось, что я не мог открыть проект после "автоматического закрытия", до тех пор, пока его не открыли на разблокированном ПК, не через удаленное подключение. На сетевом диске мы работаем каждый со своим проектом, если кто-то, что-то хочет посмотреть в твоем проекте, он просто копирует его себе локально. Открытый проект не мешает его копировать.

closed 2023-03-10 by idzm