sav-oss-eplanner/EasyEPLANner #1175: Проблема с открытием обмена сигналов для проектов типа шлюз

KUZMA-1989, 2023-06-05:
При открытии проектов, в которых есть обмен типа DO->DO, DI->DI и т.д, то не открывается межконтроллерный обмен

KirillGutyrchik, 2023-06-06:
@KUZMA-1989.
В файле описания межпроектного обмена необходимо, чтобы группы сигналов соответсвовали противоположным подтипам.
Чтобы описание прочиталось корректно, необходимо вручную в связанном проекте заменить названия груп на противоположные:
[image]
[image]

Также ссылка на документацию про ручной перевод обмена для работы в Eplanner: [13 Ручная доработка старого обмена сигналами в новый](https://github.com/savushkin-r-d/EasyEPLANner/blob/master/docs/user_manual/InterprojectExchange.md#13-Ручная-доработка-старого-обмена-сигналами-в-новый)

Также, чтобы включить в интерфейсе надстройки привязку сигналов одинакового типа (DO->DO), необходимо в параметрах фильра отключить контроль пар сигналов:
[image]

KUZMA-1989, 2023-06-06:
@

> @KUZMA-1989. В файле описания межпроектного обмена необходимо, чтобы группы сигналов соответсвовали противоположным подтипам. Чтобы описание прочиталось корректно, необходимо вручную в связанном [...]

@KirillGutyrchik На картинках описания в документации графически выделить изменения, которые нужно внести, для большей наглядности.

closed 2023-06-14 by idzm